ros-industrial/universal_robot #274: ur_kinematics fails to build on indigo

tfoote, 2016-12-29:
It looks like a missing include file.

```
00:08:02 In file included from /tmp/binarydeb/ros-indigo-ur-kinematics-1.1.6/src/ur_moveit_plugin.cpp:88:0:
00:08:02 /tmp/binarydeb/ros-indigo-ur-kinematics-1.1.6/include/ur_kinematics/ur_moveit_plugin.h:97:74: fatal error: moveit/kdl_kinematics_plugin/chainiksolver_pos_nr_jl_mimic.hpp: No such file or directory
00:08:02  #include <moveit/kdl_kinematics_plugin/chainiksolver_pos_nr_jl_mimic.hpp>
00:08:02     
```

http://build.ros.org/view/Ibin_uS64/job/Ibin_uS64__ur_kinematics__ubuntu_saucy_amd64__binary/41/console

130s, 2016-12-29:
Prerelease test against Indigo-Trusty succeeded.
```
$ mkdir /tmp/prerelease_ur
$ cd /tmp/prerelease_ur
$ generate_prerelease_script.py   https://raw.githubusercontent.com/ros-infrastructure/ros_buildfarm_config/production/index.yaml indigo default ubuntu trusty amd64  universal_robot --level 0 --custom-repo universal_robot:git:https://github.com/130s/universal_robot:fix/274   --output-dir ./
$ ./prerelease.sh 
:
# END SUBSECTION
+ echo # END SECTION
# END SECTION

Build step 8
+ [  = false ]

No test results in overlay workspace
```

130s, 2016-12-29:
Sorry, above was wrong comment. Deleted.

commit 8f7f2418 (2016-12-29): Fix #274
closed 2016-12-29 by gavanderhoorn