ros-industrial/universal_robot #281: ur_gazebo fails under Kinetic with "Action client not connected: /follow_joint_trajectory"

SimonBirrell, 2017-01-02:
Hi,

I've compiled universal_robot under Kinetic, making the modification mentioned here:
https://github.com/ros-industrial/universal_robot/issues/240

I'm attempting to run the Gazebo demo:
http://wiki.ros.org/ur_gazebo

It sort of works - Gazebo and Rviz come up but execution of any plan fails. The problem seems to be when I run 

```
roslaunch ur10_moveit_config ur10_moveit_planning_execution.launch sim:=true
```

This tries unsuccessfully to connect twice to /follow_joint_tractory then fails with 

```    
Action client not connected: /follow_joint_trajectory
```

This topic doesn't appear with rostopic list.

The output from the first terminal is

```
roslaunch ur_gazebo ur10.launch
... logging to /home/simonbirrell/.ros/log/2de0487e-d0c7-11e6-88db-000c294ec09d/roslaunch-ubuntu-43530.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
started roslaunch server http://ubuntu:33865/

SUMMARY
========

PARAMETERS
 * /arm_controller/action_monitor_rate: 10
 * /arm_controller/constraints/elbow_joint/goal: 0.1
 * /arm_controller/constraints/elbow_joint/trajectory: 0.1
 * /arm_controller/constraints/goal_time: 0.6
 * /arm_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /arm_controller/constraints/shoulder_lift_joint/trajectory: 0.1
 * /arm_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /arm_controller/constraints/shoulder_pan_joint/trajectory: 0.1
 * /arm_controller/constraints/stopped_velocity_tolerance: 0.05
 * /arm_controller/constraints/wrist_1_joint/goal: 0.1
 * /arm_controller/constraints/wrist_1_joint/trajectory: 0.1
 * /arm_controller/constraints/wrist_2_joint/goal: 0.1
 * /arm_controller/constraints/wrist_2_joint/trajectory: 0.1
 * /arm_controller/constraints/wrist_3_joint/goal: 0.1
 * /arm_controller/constraints/wrist_3_joint/trajectory: 0.1
 * /arm_controller/joints: ['shoulder_pan_jo...
 * /arm_controller/state_publish_rate: 25
 * /arm_controller/stop_trajectory_duration: 0.5
 * /arm_controller/type: position_controll...
 * /joint_state_controller/publish_rate: 50
 * /joint_state_controller/type: joint_state_contr...
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 50.0
 * /robot_state_publisher/tf_prefix: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.6
 * /use_sim_time: True

NODES
  /
[... 213 lines cut ...]
*     - CartesianPathService
*     - ExecuteTrajectoryService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1483347031.454952305, 27.853000000]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1483347031.456931657, 27.853000000]: MoveGroup context initialization complete

You can start planning now!
```

This seems similar to issue 159
https://github.com/ros-industrial/universal_robot/issues/159
and suggests that the action server has gone down for some reason. However, I don't see any error messages in the first terminal.

Any ideas? I know the Kinetic version isn't launched yet but other reasons, I really really need to move to Kinetic. Thanks!

Simon

shaun-edwards, 2017-01-02:
@SimonBirrell, thanks for reporting.  I was curious, so I tried to launch the gazebo ur sim on my system (in indigo).  I thought I'd at least be able to narrow it down to a ROS distro difference.  Unfortunately, I can't even load the xacro in indigo :(
```
shaun@shaun-Precision-5510:~/ros_industrial/universal/src/universal_robot/ur_description/urdf$ rosrun xacro xacro.py ur5_robot.urdf.xacro 
...
xacro.XacroException: Property wasn't defined: u'pi'
```

Can you determine if the action server has started and/or is running?  Running `rostopic` to find the action topics and `rosnode` would be helpful.

SimonBirrell, 2017-01-02:
Hi!

```
rosnode list
/fake_joint_calibration
/gazebo
/move_group
/robot_state_publisher
/rosout
```

and

```
rostopic list
/attached_collision_object
/calibrated
/clock
/collision_object
/execute_trajectory/cancel
/execute_trajectory/feedback
/execute_trajectory/goal
/execute_trajectory/result
/execute_trajectory/status
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/set_link_state
/gazebo/set_model_state
/joint_states
/move_group/cancel
/move_group/display_contacts
/move_group/display_planned_path
/move_group/feedback
/move_group/goal
/move_group/monitored_planning_scene
/move_group/ompl/parameter_descriptions
/move_group/ompl/parameter_updates
/move_group/plan_execution/parameter_descriptions
/move_group/plan_execution/parameter_updates
/move_group/planning_scene_monitor/parameter_descriptions
/move_group/planning_scene_monitor/parameter_updates
/move_group/result
/move_group/sense_for_plan/parameter_descriptions
/move_group/sense_for_plan/parameter_updates
/move_group/status
/move_group/trajectory_execution/parameter_descriptions
/move_group/trajectory_execution/parameter_updates
/pickup/cancel
/pickup/feedback
/pickup/goal
/pickup/result
/pickup/status
/place/cancel
/place/feedback
/place/goal
/place/result
/place/status
/planning_scene
/planning_scene_world
/rosout
/rosout_agg
/tf
/tf_static
/trajectory_execution_event
```

Thanks!

Simon

shaun-edwards, 2017-01-02:
I have a similar sim (not UR, but kuka).  It has a ` /joint_trajectory_action/goal` topic.  When i run `rostopic info`, I get:
```
Type: control_msgs/FollowJointTrajectoryActionGoal

Publishers: 
 * /my_node (http://shaun-Precision-5510:36613/)

Subscribers: 
 * /gazebo (http://shaun-Precision-5510:40339/)
```

I believe the `arm_controller_spawner` node is what loads the action into gazebo.  You should see messages similar to below, when everything is working:

```
[ INFO] [1483378160.855813772, 0.170000000]: Loading gazebo_ros_control plugin
[ INFO] [1483378160.855852267, 0.170000000]: Starting gazebo_ros_control plugin in namespace: /
[ INFO] [1483378160.856910019, 0.170000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [robot_description] on the ROS param server.
[ INFO] [1483378160.997802758, 0.170000000]: Loaded gazebo_ros_control.
[arm_controller_spawner-8] process has finished cleanly
```

gavanderhoorn, 2017-01-02:
@SimonBirrell: I can't help you with Kinetic, but I at least fixed `ur_description` for Indigo and slightly improved user experience with `ur_gazebo` and `urN_moveit_config`.

Following the instructions in the readme now works for me to start up a Gazebo sim with a UR, MoveIt and RViz with the MoveIt plugin. Planning and execution all work.

gavanderhoorn, 2017-01-02:
@shaun-edwards: if you're using 'my' Gazebo pkgs, then the setup in those is not directly comparable to the one `ur_gazebo` uses. They're similar, but not identical.

shaun-edwards, 2017-01-02:
@gavanderhoorn, do you have a PR for the `ur_description` fix?

gavanderhoorn, 2017-01-02:
No. I used my administrative access to push the fixes directly to `indigo-devel`. The current release is broken and I've a very narrow window to get a fixed version synced with the public repositories.

A brake with policy, but I deemed it acceptable in this situation.

gavanderhoorn, 2017-01-02:
See #280 for the sync to the release branch `indigo`.

shaun-edwards, 2017-01-02:
Ok...just making sure I didn't miss something.  Thanks for helping out!

gavanderhoorn, 2017-01-02:
Released in ros/rosdistro#13516.

gavanderhoorn, 2017-01-02:
This issue is somewhat of a duplicate of #231, but I've asked the OP of that issue whether we can merge it with this one.

gavanderhoorn, 2017-01-02:
@SimonBirrell wrote:
> This seems similar to issue #159 and suggests that the action server has gone down for some reason. However, I don't see any error messages in the first terminal.

This is most likely a rather different issue, as `ur_driver` is not involved when running things against Gazebo (or any simulation really).

shaun-edwards, 2017-01-02:
FYI, the joint trajectory action appears to be running in indigo with @gavanderhoorn's fix:

`rostopic list`
```
/arm_controller/command
/arm_controller/follow_joint_trajectory/cancel
/arm_controller/follow_joint_trajectory/feedback
/arm_controller/follow_joint_trajectory/goal
/arm_controller/follow_joint_trajectory/result
/arm_controller/follow_joint_trajectory/status
/arm_controller/state
```

gavanderhoorn, 2017-01-02:
There was never really a problem with the action server (it's run by `gazebo_ros_control`): the problem was that a constant used by the xacros was undefined, leading to the `ur_gazebo/urX.launch` file never completely finishing, thus not loading Gazebo or spawning the controllers. That in turn leads to the action interface never coming up, etc etc.

I don't know right now what the problem is under Kinetic, but at least on Indigo that is what happened before `1.1.9` (and since `1.1.7`).

SimonBirrell, 2017-01-03:
Hi,

I pulled 1.1.9 and did a catkin_make. The problem is still there.

I've noticed a new error message when running roslaunch ur_gazebo ur10.launch

´´´
roslaunch ur_gazebo ur10.launch
... logging to /home/simonbirrell/.ros/log/f8da7b6c-d18a-11e6-88db-000c294ec09d/roslaunch-ubuntu-68758.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

redefining global property: pi
when processing file: /home/simonbirrell/catkin_ws/src/universal_robot/ur_description/urdf/ur10.urdf.xacro
included from: /home/simonbirrell/catkin_ws/src/universal_robot/ur_description/urdf/ur10_robot.urdf.xacro
started roslaunch server http://ubuntu:44427/

SUMMARY
========

PARAMETERS
 * /arm_controller/action_monitor_rate: 10
 * /arm_controller/constraints/elbow_joint/goal: 0.1
 * /arm_controller/constraints/elbow_joint/trajectory: 0.1
 * /arm_controller/constraints/goal_time: 0.6
 * /arm_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /arm_controller/constraints/shoulder_lift_joint/trajectory: 0.1
 * /arm_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /arm_controller/constraints/shoulder_pan_joint/trajectory: 0.1
 * /arm_controller/constraints/stopped_velocity_tolerance: 0.05
 * /arm_controller/constraints/wrist_1_joint/goal: 0.1
 * /arm_controller/constraints/wrist_1_joint/trajectory: 0.1
 * /arm_controller/constraints/wrist_2_joint/goal: 0.1
 * /arm_controller/constraints/wrist_2_joint/trajectory: 0.1
 * /arm_controller/constraints/wrist_3_joint/goal: 0.1
 * /arm_controller/constraints/wrist_3_joint/trajectory: 0.1
 * /arm_controller/joints: ['shoulder_pan_jo...
 * /arm_controller/state_publish_rate: 25
 * /arm_controller/stop_trajectory_duration: 0.5
 * /arm_controller/type: position_controll...
 * /joint_state_controller/publish_rate: 50
 * /joint_state_controller/type: joint_state_contr...
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 50.0
 * /robot_state_publisher/tf_prefix: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.6
 * /use_sim_time: True

NODES
  /
    arm_controller_spawner (controller_manager/controller_manager)
    fake_joint_calibration (rostopic/rostopic)
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    joint_state_controller_spawner (controller_manager/controller_manager)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawn_gazebo_model (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [68772]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f8da7b6c-d18a-11e6-88db-000c294ec09d
process[rosout-1]: started with pid [68785]
started core service [/rosout]
process[gazebo-2]: started with pid [68793]
process[gazebo_gui-3]: started with pid [68804]
process[spawn_gazebo_model-4]: started with pid [68816]
process[robot_state_publisher-5]: started with pid [68820]
process[fake_joint_calibration-6]: started with pid [68821]
process[joint_state_controller_spawner-7]: started with pid [68823]
process[arm_controller_spawner-8]: started with pid [68836]
spawn_model script started
[INFO] [1483430552.497400, 0.000000]: Loading model xml from ros parameter
[INFO] [1483430552.508283, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[ INFO] [1483430552.687976917]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1483430552.695259646]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1483430553.959535151, 0.022000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[INFO] [1483430554.030640, 0.071000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1483430554.739896, 0.285000]: Spawn status: SpawnModel: Successfully spawned model
[ INFO] [1483430554.819791086, 0.285000000]: Physics dynamic reconfigure ready.
Unhandled exception in thread started by 
sys.excepthook is missing
lost sys.stderr
[spawn_gazebo_model-4] process has finished cleanly
log file: /home/simonbirrell/.ros/log/f8da7b6c-d18a-11e6-88db-000c294ec09d/spawn_gazebo_model-4*.log
´´´

It's towards the end:

´´´
Unhandled exception in thread started by 
sys.excepthook is missing
lost sys.stderr
´´´

That wasn't there when I opened the issue, but I'm not sure when it started appearing.

It's clear that arm_controller isn't loading for some reason. I'm guessing what to try next, but I've tried:

1. Running roslaunch ur_gazebo ur10.launch in one terminal
2. Launching the arm_controller_spawner manually with:

```
rosrun controller_manager controller_manager spawn arm_controller respawn:="false" output:="screen"
```

I get no output at all, but this is probably the wrong way to go about things.

What can I try next to track down the issue?

Thanks

Simon

shaun-edwards, 2017-01-03:
Do you know which node is throwing the unhandled exception?  Once fully loaded, you should see the following nodes:
```
shaun@shaun-Precision-5510:~/ros_industrial/universal$ rosnode list
/fake_joint_calibration
/gazebo
/robot_state_publisher
/rosout
```
In addition to the terminal output, you should see the Gazebo sim window, like below:
[image]

SimonBirrell, 2017-01-04:
Here's what the screen looks like after running 

```
roslaunch ur_gazebo ur10.launch
```

[image]

So I have the same nodes as as you. Today I didn't get the exception. Go figure. I then run the second terminal:

```
roslaunch ur10_moveit_config ur10_moveit_planning_execution.launch sim:=true
```

[image]

The same nodes still, no sign of arm_controller.

Simon

SimonBirrell, 2017-01-04:
Are we aware of anybody who has this working on Kinetic? Or anyone else who has tried?

shaun-edwards, 2017-01-04:
@SimonBirrell, I know people are using Kinetic, but I'm not aware of anyone who is using the universal gazebo sim.  I recommend and email to the users group (swri-ros-pkg-dev@googlegroups.com).  There a number of users of universal who might be able to help.  Sorry this isn't working for you.

SimonBirrell, 2017-01-06:
FOUND IT!

The solution is absurdly simple, of course. You need to reinstall the gazebo_ros_pkgs. Presumably the latest version isn't included in the version of Gazebo that comes with Kinetic.

So just type

```
sudo apt-get install ros-kinetic-gazebo-ros-pkgs ros-kinetic-gazebo-ros-control
```

after installing Kinetic and all the bits and pieces listed on http://wiki.ros.org/ur_gazebo and applying fix #240 . 
More details at http://gazebosim.org/tutorials?tut=ros_installing

I'm still having some issues executing paths correctly, but this is clearly a separate issue. All /arm_controller/follow_joint_trajectory topics are now present and correct.

It seems to me you may have a functioning Kinect release once you accept #240. Might be worth taking the plunge.

Thanks for all the help and advice.

closed 2017-01-06 by SimonBirrell

SimonBirrell, 2017-01-06:
A final note for the lost souls who find this. I also needed to adjust the parameter longest_valid_segment_fraction to be 0.02 in ur10_moveit_config/config/ompl_planning.yaml
as detailed in 
https://groups.google.com/forum/#%21topic/moveit-users/zNRG50tHuuY

Substitute ur5 for ur10 as necessary. 

With that, I have a fully-working Gazebo demo of the UR10 under Kinetic!