ros-industrial/universal_robot #287: Trajectory time exceeded and current robot state not at goal, last point required

XeMinZa, 2017-01-06:
When I used ur_driver and MoveIt to control real UR5, the UR5 can move but can't reach to the goal pose.In the ur_bringup terminal ,an [WARN] occurred
[2016-12-29 22:35:28.417916] on_goal
[WARN] [WallTime: 1483022131.205065] Trajectory time exceeded and current robot state not at goal, last point required
[WARN] [WallTime: 1483022131.205451] Current trajectory time: 2.78684282303, last point time: 2.779444779

 Desired: [0.42070332844406366, -0.5392611878419296, 1.2213945445207879, -0.014400772983208299, -0.02953762591350824, 0.020142455794848496]

actual: [0.40468594431877136, -0.7387240568744105, 1.479959487915039, -0.6717637220965784, 0.7213889360427856, -0.06885129610170537]

velocity: [0.013294500298798084, 0.14815698564052582, -0.19076329469680786, 0.4873025715351105, -0.5491651296615601, 0.056661415845155716]

[2016-12-29 22:35:31.724464] Out: Braking
After running the ur5_bringup.launch , my move_group.launch(generated by moveit_setup_assistant),and the moveit_ik_demo.py,The UR5 begins to move toward the desired pose.But it came to a sudden stop in the journey .and can't reach to the desired pose. The warning is "Trajectory time exceeded and current robot state not at goal, last point required". I do not know how to set trajectory time.I also don't know the reason that caused the problem.

shaun-edwards, 2017-01-06:
Can you tell us what versions you are using?  Also, how close to the goal are you getting?  Have you followed the directions in the [README](https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/README.md)?

XeMinZa, 2017-01-07:
Polyscope 3.1.18024,and I am using Ubuntu 14.04,ROS Indigo.When I run the demo based on inverse kinematics(moveit_ik_demo.py),It had a braking at the point very close to the goal.I have read the directions many times.I added a desk to the urdf of UR5,and the moveit_config is generated according to the new urdf.My moveit_ik_demo.py can be found [here](http://answers.ros.org/question/250790/using-universal_robot-pkg-and-moveit-to-control-real-ur5-robotur5-can-move-but-cant-move-to-the-desired-poseit-saied-time-exceeded-and-current-robot/).  When I used moveit_fk_demo.py,which is based on forward kinematics，the stopping point is a little far to the goal.After the stopping,It all appeared the similar warning.Thanks a lot

gavanderhoorn, 2018-06-25:
I do not see any info about which specific driver was being used @XeMinZa.

If this is still an issue for you -- and you are not using `ur_driver` -- then please open an issue on the tracker of one of the replacement drivers.

`ur_driver` is deprecated, certainly for CB3 controllers.

gavanderhoorn, 2018-06-25:
Closing due to no activity and because this doesn't seem like the appropriate tracker.

closed 2018-06-25 by gavanderhoorn