ros-industrial/universal_robot #288: Failed converting of xacro to URDF

Chunting, 2017-02-18:
I converted ur5.urdf.xacro to ur5.urdf, but it does not work. the result is shown as below.
The command I run is **rosrun xacro xacro.py ur5.urdf.xacro > ur5.urdf**
The ur5.urdf.xacro is from the [universal_robot,](https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/ur_description/urdf/ur5.urdf.xacro) and I do not think it has a problem.

Would you please tell why it happens and how can I make it work in rviz?
```
<?xml version="1.0" ?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from ur5.urdf.xacro                 | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">
  <!-- Inertia parameters -->
  <!-- This mass might be incorrect -->
  <!-- Kinematic model -->
  <!-- Properties from urcontrol.conf -->
  <!--
    DH for UR5:
    a = [0.00000, -0.42500, -0.39225,  0.00000,  0.00000,  0.0000]
    d = [0.089159,  0.00000,  0.00000,  0.10915,  0.09465,  0.0823]
    alpha = [ 1.570796327, 0, 0, 1.570796327, -1.570796327, 0 ]
    q_home_offset = [0, -1.570796327, 0, -1.570796327, 0, 0]
    joint_direction = [-1, -1, 1, 1, 1, 1]
    mass = [3.7000, 8.3930, 2.2750, 1.2190, 1.2190, 0.1879]
    center_of_mass = [ [0, -0.02561, 0.00193], [0.2125, 0, 0.11336], [0.11993, 0.0, 0.0265], [0, -0.0018, 0.01634], [0, 0.0018,0.01634], [0, 0, -0.001159] ]
  -->
  <!-- Arbitrary offsets for shoulder/elbow joints -->
  <!-- measured from model -->
  <!-- measured from model -->
  <!-- link lengths used in model -->
  <!--property name="shoulder_height" value="0.089159" /-->
  <!--property name="shoulder_offset" value="0.13585" /-->
  <!-- shoulder_offset - elbow_offset + wrist_1_length = 0.10915 -->
  <!--property name="upper_arm_length" value="0.42500" /-->
  <!--property name="elbow_offset" value="0.1197" /-->
  <!-- CAD measured -->
  <!--property name="forearm_length" value="0.39225" /-->
  <!--property name="wrist_1_length" value="0.093" /-->
  <!-- CAD measured -->
  <!--property name="wrist_2_length" value="0.09465" /-->
  <!-- In CAD this distance is 0.930, but in the spec it is 0.09465 -->
  <!--property name="wrist_3_length" value="0.0823" /-->
  <!-- manually measured -->
</robot>
```

[image]

de-vri-es, 2017-02-18:
The file `ur5.urdf.xacro` defines a macro but does not invoke the macro. You can instead use `ur5_robot.urdf.xacro`, or use that as a template for your own URDF.

You should probably write your own URDF at some point to add parts of the environment and an end-effector. But for testing you could use `ur5_robot.urdf.xacro` directly.

gavanderhoorn, 2017-02-18:
As @de-vri-es explained, `ur5.urdf.xacro` is a not a *top-level* xacro, but a file defining just the `ur5_robot` xacro macro. Such files cannot be used directly, as the macro is then never called, resulting in an empty file as you already found out.

Converting `ur5_robot.urdf.xacro` should work, as that is a top-level xacro.

gavanderhoorn, 2017-02-18:
I'm closing this as a `works-for-me`.

If you need any more assistance or feel we didn't address your question, please re-open.

closed 2017-02-18 by gavanderhoorn

Chunting, 2017-02-18:
Thank you all and this is indeed the cause.