ros-industrial/universal_robot #291: Not able to open socket socket 0 at host: X.X.X.X at port 50001

AAvi, 2017-03-02:
Hey,

Help please!!!

The error in the title showed up in the UR10 control panel info while trying to send a goal to the arm (such as in test_move.py).
 
details:
- UR package from source, and also the ur_modern_driver (same issue with ur_driver)
-UR10 firmware version 3.2 on CB3, same issue with versions 3.0 and 3.1. 
- Ubuntu 14.04, Indigo.

We are able to update the payload without any error. Also $rostopic hz is 125 Hz for arm topics. It should be noted that there were no controllers found running : $rosservice call /controller_manager/list_controllers {}

Changed the port though ur_ros_wrapper.cpp - same issue with different port from 50001

Thank you very much in advance for helping with this problem.

gavanderhoorn, 2017-03-02:
Please tell us *exactly* how you start the driver, which launch files, in which order, etc.

Further: are you running this in a Virtual Machine? If not, does your computer have multiple network interfaces? Do you have a firewall enabled?

> Changed the port though ur_ros_wrapper.cpp - same issue with different port from 50001

don't do that, this is all configurable using the launch parameters supported by the launch files, so don't go and change any source files.

PS: "Help please!!!" is not needed. We'll be notified of your issue as soon as you open it, and we'll respond as soon as we have time.

AAvi, 2017-03-02:
Thank you for the fast response!

The following lines were suppose to work with the said hardware and software versions:
$roscore
$roslaunch ur_modern_driver ur10_bringup.launch robot_ip:=IP_OF_THE_ROBOT  reverse_port:=REVERSE_PORT]
$rosrun ur_modern_driver test_move.py

It seems that changing the "reverse_port" in the terminal did not effect the port number showed by the control panel of the robot. Only by editing ur_ros_wrapper.cpp the port actually changed (just trying to isolate the issue).

Thanks again

gavanderhoorn, 2017-03-02:
First: those are not copy-pastes from your terminal. Especially the exact form of the `roslaunch` of `ur10_bringup.launch` is important. Could you copy-paste an example of that please (you can change the IP if you want)?

Something to try:

 1. start everything as you normally do
 2. in a separate terminal, execute `netstat -antu` and copy-paste the output here (you may again change the IPs, but make sure to do it **consistenly** (ie: use search-replace))

AAvi, 2017-03-02:
Unfortunately I wont be able to get near the robot for the next few days.
These lines are exact expect the IP and Port.  The IP chosen is 192.168.1.10 and port 50001 as an example.  Tried it with and without the reverse_port command. The IP was set manually when the cross-cable was used and set automatically when connected through the network.  
by the way- same issue when sending commands using rviz (obviously)

Its hard to imagine that this is a network issue.  I'll add the output of netstat -antu as soon as I can.

In the meantime, any ideas?

Thanks again.

gavanderhoorn, 2017-03-02:
No point in speculating until we know more about your network setup / open ports.

Note that you don't need to supply the `reverse_port` argument. It's got a good default already.

AAvi, 2017-03-02:
Thank you gavanderhoorn!
I love speculations, especially with problem solving. I googled netstat -antu and figured no other option then the firewall. Loaded the Ubuntu and yes, it was on ;-| . It was odd since through the ports 3001-3002 it was able to send commands to the robot (changing the payload indeed updated the control panel payload configuration) so didn't even think to recheck the firewall's status. Moreover, on execution of test_move.py, the control panel printed out "Program driverProg started" before the error massage. 

In Linux the port numbers are divided into three ranges:
Well Known Ports: those from 0 through 1023.
Registered Ports: those from 1024 through 49151
Dynamic and/or Private Ports: those from 49152 through 65535
(https://www.cyberciti.biz/faq/linux-unix-open-ports/)

That is why I tried changing the port number under 49000 (3999, which didnt work with reverse_port btw), the same category of the ports that did work (30001-30003).

How come only some of the TCP/IP packets can be passed through the firewall? 

Resolved (not verified but... it is).  Thank you again for helping with this issue.

gavanderhoorn, 2017-03-03:
> I love speculations, especially with problem solving

sure, but only based on facts. And those were missing.

> I googled netstat -antu and figured no other option then the firewall. Loaded the Ubuntu and yes, it was on

ok, well that's your problem then.

> It was odd since through the ports 3001-3002 it was able to send commands to the robot 

well, that is not totally unexpected: (consumer) firewalls typically allow traffic *out* of the machine by default, but block everything coming from the outside.

> Moreover, on execution of test_move.py, the control panel printed out "Program driverProg started" before the error massage.

Again, not so strange, as that is uploaded *from* the PC *to* the controller.

> That is why I tried changing the port number under 49000 (3999, which didnt work with reverse_port btw), the same category of the ports that did work (30001-30003).

There is no reason to change that port at all (unless you have something already using `50001` or are setting up a multi-arm config), and whether it's an ephemeral port or not does not matter either.

And `30001` and `30002` are ports that are opened *on the controller*, not on your PC.

> How come only some of the TCP/IP packets can be passed through the firewall?

I believe this should now be clear.

gavanderhoorn, 2017-03-03:
I'm closing this as I have the impression that you've solved your issue.

Please re-open if you have any further problems (with this same issue, otherwise open a new issue).

closed 2017-03-03 by gavanderhoorn

taylorty, 2017-03-27:
I encountered a similar issue but I resolved it by resetting the firewall. Hope that could help anyone who may encounter a similar problem.