ros-industrial/universal_robot #292: OMPL about UR

lkj1114889770, 2017-03-07:
with the ur3_config and it's urdf provided, i just use command 

> roslaunch ur3_moveit_config demo.launch

to see the behavior of ur3 in the rviz. But it behaves very strange when i set a random position to get it to plan motion. I chosed unspecified for OMPL. It's always failed for motion plan, and the ur3 in the rviz always shows a very strage trajectory. I am a fresh for ros and moveit, i don't konw how to solve the problem. Please, let's me the reason perhaps for this ,and what should I do. This may help me to get a quick start for ros a and moveit. Thanks.

My ros version: kinetic
OS: Ubuntu 16.04

gavanderhoorn, 2017-03-07:
> I chosed unspecified for OMPL.

I'm assuming you mean that you did not specify a controller when submitting your planning request.

The UR robots are known to not work very well with the default planner (typically `LBKPIECE1`), so you might want to select a different one. Try the `RRTConnect` planner.

gavanderhoorn, 2017-03-07:
Btw: could I please ask you to post these kind of questions on [ROS Answers](http://answers.ros.org)?

This tracker is really only for problems with the code or any of the packages in this repository, and your question is more about how they are used with MoveIt (an external application).

lkj1114889770, 2017-03-08:
Thanks， the problem has been solved. It was due to the limitation of the joint so it seems that the robot is always being in collision. Also every time to start motion, it would be better to update start sate.

closed 2017-03-08 by lkj1114889770