ros-industrial/universal_robot #293: python interface problem

MiaoLi, 2017-03-11:
When I run the test_analytical_ik.py file, it game me a error:
"ImportError: No module named ur_kin_py"
I have tried different ways to build the module, there is no test example for this.
Any idea?

gavanderhoorn, 2017-03-11:
How did you setup your workspace? Did you source the `setup.bash` (just making sure).

I must admit I've never had a need to use `ur_kin_py` so I don't have any experience troubleshooting it.

That said this looks like a `PYTHONPATH` issue.

MiaoLi, 2017-03-14:
Finally, I made it work.
It is not the workspace problem. We need to change the CMAKELIST to make it work, also the PYTHONPATH as mentioned above.
I would suggest to either remove this file or add some tutorial doc.

gavanderhoorn, 2017-03-14:
Could you tell us what you changed?

MiaoLi, 2017-03-15:
I add the following lines in CMakeList.txt.
PYTHON_ADD_MODULE(ur_kin_py src/ur_kin_py.cpp)
target_link_libraries(ur_kin_py ur3_kin boost_python boost_numpy ${PYTHON_LIBRARIES} ${Boost_LIBRARIES} )

Build with ccatkin_make and then copy ur_kin_py.so to the folder of test_analytical_ik.py.

atenpas, 2017-05-30:
Adding what @MiaoLi suggested to _CMakeList.txt_ in _ur_kinematics_ did not produce an so file for me but a file without extension. Renaming and copying it to the folder of _test_analytical_ik.py_ allows to run _test_analytical_ik.py_. However, this results in another bug:

`File "test_analytical_ik.py", line 39, in test_q
    print i1-3, i2-3, i3-3, i4-3, i5-3, i6-3
NameError: global name 'i1' is not defined`

See https://github.com/ros-industrial/universal_robot/blob/indigo-devel/ur_kinematics/src/ur_kinematics/test_analytical_ik.py#L38.

gavanderhoorn, 2018-06-25:
Tagging as `wrid18` and adding to the `IK` project board so this can be tracked as one of the IK related issues.

ipa-nhg, 2018-07-11:
Solved by https://github.com/ros-industrial/universal_robot/pull/364

gavanderhoorn, 2018-07-29:
Fixed by #364.

closed 2018-07-29 by gavanderhoorn