ros-industrial/universal_robot #294: "Waiting for /follow_joint_trajectory to come up" and "Action client not connected"

WangJIankun1992, 2017-03-20:
I try to use gazebo simulation with ur3( ROS Kinetic,Ubuntu 16.04, ur_modern_driver, Gazebo 7 ), but I meet some problems as follows:

[ INFO] [1489997728.462448878, 756.472000000]: Waiting for /follow_joint_trajectory to come up
[ INFO] [1489997733.475260715, 761.472000000]: Waiting for /follow_joint_trajectory to come up
[ERROR] [1489997738.491130260, 766.472000000]: Action client not connected: /follow_joint_trajectory
[ INFO] [1489997738.547350281, 766.527000000]: Returned 0 controllers in list
[ INFO] [1489997738.585389624, 766.562000000]: Trajectory execution is managing controllers

I wonder that we should add the driver into the ur3_moveit_planning_execution.launch ?

gavanderhoorn, 2017-03-20:
The driver is not used when simulating everything using Gazebo.

Two things to check:

 1. did you follow the instructions in the readme?
 1. do you have `gazebo_ros_control` installed?

WangJIankun1992, 2017-03-20:
@gavanderhoorn I have installed `ros-kinetic-gazebo-ros-control`, and I followed the instructions as follows:
roslaunch ur_gazebo ur3.launch limited:=true
roslaunch ur3_moveit_config ur3_moveit_planning_execution.launch sim:=true limited:=true
roslaunch ur3_moveit_config moveit_rviz.launch config:=true
In addition, I can test on real robot(UR3) successfully.
I found when I launched ur3.launch, there were topics as follows:
/arm_controller/command
/calibrated
/clock
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/set_link_state
/gazebo/set_model_state
/joint_states
/rosout
/rosout_agg
/tf
/tf_static
But when I launched ur_modern_driver ur_bringup.lanuch, there were topics as follows:
/follow_joint_trajectory/cancel
/follow_joint_trajectory/feedback
/follow_joint_trajectory/goal
/follow_joint_trajectory/result
/follow_joint_trajectory/status
/joint_states
/rosout
/rosout_agg
/tf
/tf_static
/tool_velocity
/ur_driver/URScript
/ur_driver/io_states
/ur_driver/joint_speed
/wrench
So I am afraid th topic `follow_joint_trajectory` may be lost.

gavanderhoorn, 2017-03-20:
This most likely just means that you have something not installed.

Can you check you have `ros-kinetic-joint-trajectory-controller` installed?

WangJIankun1992, 2017-03-20:
When I run `rosdep install --from-paths /path/to/where/universal_robot/pkgs/are --ignore-src` for pkgs `ur_gazebo`, it reminds me to install `ros-kinetic-ros-controllers` and I have installed it and run the command for pkgs `ur3_moveit_config` and it reminds me nothing to install. And I have also installed `ros-kinetic-joint-trajectory-controller`, but the same problem still exists.
Could you provide me an email address so that I can send you the source code if you have free time to read ? My email address is in my Github homepage. Thank you very much.

gavanderhoorn, 2017-03-20:
> Could you provide me an email address so that I can send you the source code if you have free time to read ? My email address is in my Github homepage. Thank you very much.

no, that's not how this works.

> When I run `rosdep install --from-paths /path/to/where/universal_robot/pkgs/are --ignore-src` for pkgs `ur_gazebo`, it reminds me to install `ros-kinetic-ros-controllers` and I have installed it and [...]

Please just run `rosdep install ..` for your entire workspace, or at least for the top-level `universal_robot` folder in your workspace.

WangJIankun1992, 2017-03-20:
jiankun@jiankun:~$ rosdep install --from-paths ~/catkin_modern/src/universal_robot-kinetic-devel --ignore-src
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ur_bringup: Cannot locate rosdep definition for [ur_driver]
jiankun@jiankun:~$ rosdep install --from-paths ~/catkin_modern/src/universal_robot-kinetic-devel --ignore-src
#All required rosdeps installed successfully

Before the second command, I just remove the directory `ur_bringup` because I use `ur_modern_driver` then `ur_bringup` is not needed any more.
But the problem is not solved.

WangJIankun1992, 2017-03-20:
jiankun@jiankun:~$ rosdep install --from-paths ~/catkin_modern/src/universal_robot-kinetic-devel --ignore-src
#All required rosdeps installed successfully

jiankun@jiankun:~$ rosdep install --from-paths  ~/catkin_modern --ignore-src
#All required rosdeps installed successfully

Before the command, I just remove the directory `ur_bringup` because I use `ur_modern_driver` then `ur_bringup` is not needed any more.
But the problem is not solved.

gavanderhoorn, 2017-03-20:
How did you install ROS? From `.deb`s, or from sources? Is the Gazebo version you have the one 'packaged' by ROS, or did you install it using the instructions on `gazebosim.org`?

Basically the problem is that the Action server that exposes the `FollowJointTrajectory` action is not running. That either means that `gazebo_ros_control` is not installed (for your particular version of Gazebo) or one of the plugins cannot be found (again: by Gazebo).

Can you copy-paste the entire console output of a run of `roslaunch ur_gazebo ur3.launch limited:=true`?

WangJIankun1992, 2017-03-20:
jiankun@jiankun:~$ roslaunch ur_gazebo ur3.launch limited:=true
... logging to /home/jiankun/.ros/log/9a4ee926-0d57-11e7-a66e-208984d70fef/roslaunch-jiankun-9855.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

redefining global property: pi
when processing file: /home/jiankun/catkin_modern/src/universal_robot-kinetic-devel/ur_description/urdf/ur3.urdf.xacro
included from: /home/jiankun/catkin_modern/src/universal_robot-kinetic-devel/ur_description/urdf/ur3_joint_limited_robot.urdf.xacro
started roslaunch server http://jiankun:41353/

SUMMARY
========

PARAMETERS
 * /arm_controller/action_monitor_rate: 10
 * /arm_controller/constraints/elbow_joint/goal: 0.1
 * /arm_controller/constraints/elbow_joint/trajectory: 0.1
 * /arm_controller/constraints/goal_time: 0.6
 * /arm_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /arm_controller/constraints/shoulder_lift_joint/trajectory: 0.1
 * /arm_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /arm_controller/constraints/shoulder_pan_joint/trajectory: 0.1
 * /arm_controller/constraints/stopped_velocity_tolerance: 0.05
 * /arm_controller/constraints/wrist_1_joint/goal: 0.1
 * /arm_controller/constraints/wrist_1_joint/trajectory: 0.1
 * /arm_controller/constraints/wrist_2_joint/goal: 0.1
 * /arm_controller/constraints/wrist_2_joint/trajectory: 0.1
 * /arm_controller/constraints/wrist_3_joint/goal: 0.1
 * /arm_controller/constraints/wrist_3_joint/trajectory: 0.1
 * /arm_controller/joints: ['shoulder_pan_jo...
 * /arm_controller/state_publish_rate: 25
 * /arm_controller/stop_trajectory_duration: 0.5
 * /arm_controller/type: position_controll...
 * /joint_state_controller/publish_rate: 50
 * /joint_state_controller/type: joint_state_contr...
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 50.0
 * /robot_state_publisher/tf_prefix: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.6
 * /use_sim_time: True

NODES
  /
    arm_controller_spawner (controller_manager/controller_manager)
    fake_joint_calibration (rostopic/rostopic)
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    joint_state_controller_spawner (controller_manager/controller_manager)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawn_gazebo_model (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [9869]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 9a4ee926-0d57-11e7-a66e-208984d70fef
process[rosout-1]: started with pid [9882]
started core service [/rosout]
process[gazebo-2]: started with pid [9900]
process[gazebo_gui-3]: started with pid [9911]
process[spawn_gazebo_model-4]: started with pid [9916]
process[robot_state_publisher-5]: started with pid [9917]
process[fake_joint_calibration-6]: started with pid [9918]
process[joint_state_controller_spawner-7]: started with pid [9928]
process[arm_controller_spawner-8]: started with pid [9943]
SpawnModel script started
[INFO] [1490005557.888684, 0.000000]: Loading model XML from ros parameter
[INFO] [1490005557.894534, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[ INFO] [1490005557.945411013]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1490005557.951461817]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1490005558.455026370, 0.023000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[INFO] [1490005558.501680, 0.067000]: Calling service /gazebo/spawn_urdf_model
[ INFO] [1490005558.511134883, 0.076000000]: Physics dynamic reconfigure ready.
[INFO] [1490005558.694283, 0.126000]: Spawn status: SpawnModel: Successfully spawned model
[ INFO] [1490005558.826759903, 0.126000000]: Loading gazebo_ros_control plugin
[ INFO] [1490005558.826933143, 0.126000000]: Starting gazebo_ros_control plugin in namespace: /
[ INFO] [1490005558.828593700, 0.126000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [robot_description] on the ROS param server.
[spawn_gazebo_model-4] process has finished cleanly
log file: /home/jiankun/.ros/log/9a4ee926-0d57-11e7-a66e-208984d70fef/spawn_gazebo_model-4*.log
[ INFO] [1490005559.008405435, 0.126000000]: Loaded gazebo_ros_control.
Loaded joint_state_controller
Started ['joint_state_controller'] successfully
[joint_state_controller_spawner-7] process has finished cleanly
log file: /home/jiankun/.ros/log/9a4ee926-0d57-11e7-a66e-208984d70fef/joint_state_controller_spawner-7*.log
Loaded arm_controller
Started ['arm_controller'] successfully
[arm_controller_spawner-8] process has finished cleanly
log file: /home/jiankun/.ros/log/9a4ee926-0d57-11e7-a66e-208984d70fef/arm_controller_spawner-8*.log

gavanderhoorn, 2017-03-20:
I've just installed `universal_robot` in a clean Kinetic Docker container, then started:

```
$ roslaunch ur_gazebo ur3_joint_limited.launch
```

a `rostopic list` gives me:

```
/arm_controller/command
/arm_controller/follow_joint_trajectory/cancel
/arm_controller/follow_joint_trajectory/feedback
/arm_controller/follow_joint_trajectory/goal
/arm_controller/follow_joint_trajectory/result
/arm_controller/follow_joint_trajectory/status
/arm_controller/state
/calibrated
/clock
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/set_link_state
/gazebo/set_model_state
/joint_states
/rosout
/rosout_agg
/tf
/tf_static
```

This would seem to be all expected and required topics, including the action server.

Steps I used to setup my workspace:

```bash
$ mkdir -p /tmp/ws/src
$ cd /tmp/ws/src
$ git clone https://github.com/ros-industrial/universal_robot.git
$ cd /tmp/ws
$ rosdep update
$ rosdep install --from-paths src --ignore-src
$ catkin_make
$ source devel/setup.bash
```

At this point I start `roslaunch ur_gazebo ur3_joint_limited.launch`.

Can you try the above steps in a clean terminal (no workspace `setup.bash` `source`d) and a clean workspace?

WangJIankun1992, 2017-03-20:
WOW! Thank you very much. According to your suggestion, now the real robot and gazebo simulation can work well separately. 
I think my problem is about the link between ROS and gazebo. Because originally I installed ROS Kinetic and gazebo7 separately, then I installed some pkgs such as you mentioned above. So maybe there is something lost about the link between ros and gazebo. And now I install ros using the recommended method `sudo apt-get install ros-kinetic-desktop-full`, then the problem disappear.
Thanks again for your help. I am very interested in ROS-Industrial and I am a PhD student managing robot control and trajectory planning in the Chinese University of Hong Kong. I hope that I can go to Delft University of Technology for an intern or exchange to study the robot knowledge if possible in the future. So I wonder if there is any position?

gavanderhoorn, 2017-03-20:
Good to hear you got it to work. Can you please close the issue if you feel it has been resolved?

re: internships: at the moment we don't have any open positions, but you can always try at a later time.

WangJIankun1992, 2017-03-20:
OK.I will close it. By the way, could you give me an email address or website where I can follow the latest information about internships?

closed 2017-03-20 by WangJIankun1992