ros-industrial/universal_robot #295: Change material color of gazebo model

jacknlliu, 2017-03-23:
Could we change material color  of gazebo model to make it more like V-REP style, rather than LightGrey for every link?

The UR model in V-REP is more closer to its real color.

shaun-edwards, 2017-03-23:
Can you provide images of the current and desired robot model?

jacknlliu, 2017-03-23:
@shaun-edwards  still no available robot model

shaun-edwards, 2017-03-23:
I mean can you provide an example of the color difference (i.e. the current color in ROS/Gazebo) vs. the desired color in V-REP?

jacknlliu, 2017-03-23:
@shaun-edwards  I can't get the material color RGB value of UR model from V-REP.  I get a screenshot from V-REP to show the difference.

UR5 V-REP model screenshot:
[image]

UR5 Gazebo model screenshot:
[image]

gavanderhoorn, 2017-03-23:
I think this is more a difference in lighting between V-REP and RViz than a difference in material settings.

jacknlliu, 2017-03-23:
@gavanderhoorn  I don't think it's about lighting.  If you run `roslaunch gazebo_ros rubble_world.launch` , you will find the lighting is the same as UR settings, but the visual effect is different.

ipa-nhg, 2018-07-11:
Solved by https://github.com/ros-industrial/universal_robot/pull/359
[image]

closed 2018-07-11 by ipa-nhg