ros-industrial/universal_robot #296: moveit planning always fail

jacknlliu, 2017-04-12:
I use gazebo simulation with UR5 with the following commands,
```
$ roslaunch ur_gazebo ur5.launch limited:=true

$ roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true limited:=true

$ roslaunch ur5_moveit_config moveit_rviz.launch config:=true
```
But moveit planning with interactive marker always fail.

I use the `kinetic-devel`branch.

rahulbish, 2017-04-18:
i have the same problem

gavanderhoorn, 2017-04-25:
I haven't had a chance to reproduce this, but in the meantime: could you please show us some of the console output that you see when the planning fails? I'm particularly interested in the MoveIt output where it instantiates the planners and sets them to work.

jacknlliu, 2017-04-25:
I found that this issue is about my robot tools configuration which has a collision, then moveit detected the collision, and planning failed.

Should I close this issue? This issue seems not about this package.

closed 2017-04-25 by jacknlliu

gavanderhoorn, 2017-04-25:
You already closed it, but indeed: if this is not related to the packages in this repository then we should close the issue.

Thanks for your response. Seems everything was working as it should.