ros-industrial/universal_robot #297: real UR5 robot did not move with moveit! planning

jacknlliu, 2017-05-05:
I used the following command to drive the real UR5 robot with moveit_rviz in three terminal windows:
```
$ roslaunch ur_bringup ur5_bringup.launch limited:=true robot_ip:="192.168.1.125"
$ roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch limited:=true
$ roslaunch ur5_moveit_config moveit_rviz.launch config:=true
```
then I make robot planning with `moveit_rviz`, but the UR5 robot did not move even the `plan` succeed every time.

I used the `kinetic-devel` branch.

The following was what I got:

- startup
```
$ roslaunch ur_bringup ur5_bringup.launch limited:=true robot_ip:="192.168.1.125"

Setting prefix to 
[WARN] [1493953176.316694]: No calibration offset for joint "shoulder_pan_joint"
[WARN] [1493953176.316989]: No calibration offset for joint "shoulder_lift_joint"
[WARN] [1493953176.317182]: No calibration offset for joint "elbow_joint"
[WARN] [1493953176.317359]: No calibration offset for joint "wrist_1_joint"
[WARN] [1493953176.317531]: No calibration offset for joint "wrist_2_joint"
[WARN] [1493953176.317699]: No calibration offset for joint "wrist_3_joint"
[INFO] [1493953176.317888]: No calibration offsets loaded from urdf
[INFO] [1493953176.319197]: Max velocity accepted by ur_driver: 10.0 [rad/s]
[INFO] [1493953176.321557]: Bounds for Payload: [0.0, 10.0]
Disconnected.  Reconnecting
[INFO] [1493953176.350491]: Programming the robot
[INFO] [1493953176.353631]: Programming the robot at 192.168.1.125
Likely a syntax error:

Exception in thread URConnection:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/data/UR5/catkin_ws/src/ur_driver/src/ur_driver/driver.py", line 274, in __run
    self.__on_packet(packet)
  File "/data/UR5/catkin_ws/src/ur_driver/src/ur_driver/driver.py", line 186, in __on_packet
    state = RobotState.unpack(buf)
  File "/data/UR5/catkin_ws/src/ur_driver/src/ur_driver/deserialize.py", line 324, in unpack
    raise Exception("Fatal error when unpacking RobotState packet")
Exception: Fatal error when unpacking RobotState packet

[INFO] [1493953177.364591]: Programming the robot at 192.168.1.125
[INFO] [1493953178.374196]: Programming the robot at 192.168.1.125
[INFO] [1493953179.383268]: Programming the robot at 192.168.1.125
[INFO] [1493953180.395221]: Programming the robot at 192.168.1.125
[INFO] [1493953181.404775]: Programming the robot at 192.168.1.125
[INFO] [1493953182.414408]: Programming the robot at 192.168.1.125
...
```

- planning
```
$  roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch limited:=true

[ INFO] [1493953185.138356131]: Using planning request adapter 'Fix Start State Path Constraints'
[ WARN] [1493953185.141752927]: 
Deprecation warning: parameter 'allowed_execution_duration_scaling' moved into namespace 'trajectory_execution'.
Please, adjust file trajectory_execution.launch.xml!
[ WARN] [1493953185.142617127]: 
Deprecation warning: parameter 'allowed_goal_duration_margin' moved into namespace 'trajectory_execution'.
Please, adjust file trajectory_execution.launch.xml!
```
```
[ INFO] [1493953190.194856980]: Waiting for /follow_joint_trajectory to come up
[ INFO] [1493953195.195041719]: Waiting for /follow_joint_trajectory to come up
[ERROR] [1493953200.195360835]: Action client not connected: /follow_joint_trajectory
[ INFO] [1493953200.302668376]: Returned 0 controllers in list
[ INFO] [1493953200.332978079]: Trajectory execution is managing controllers
```
```
Loading 'move_group/ApplyPlanningSceneService'...
Loading 'move_group/ClearOctomapService'...
Loading 'move_group/MoveGroupCartesianPathService'...
Loading 'move_group/MoveGroupExecuteService'...
Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
Loading 'move_group/MoveGroupGetPlanningSceneService'...
Loading 'move_group/MoveGroupKinematicsService'...
Loading 'move_group/MoveGroupMoveAction'...
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...
[ INFO] [1493953200.506885369]: 

********************************************************
* MoveGroup using: 
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1493953200.507007962]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1493953200.507051218]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1493953232.190879270]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1493953232.192702691]: Planner configuration 'manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1493953232.193629587]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1493953232.193806414]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1493953232.193866423]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1493953232.193907965]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1493953232.206145231]: RRTConnect: Created 5 states (3 start + 2 goal)
[ INFO] [1493953232.207005216]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1493953232.208742147]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1493953232.214805545]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1493953232.215104491]: ParallelPlan::solve(): Solution found by one or more threads in 0.021817 seconds
[ INFO] [1493953232.215410686]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1493953232.215483549]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1493953232.215882385]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1493953232.216119737]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1493953232.216454975]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1493953232.216766575]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1493953232.217223672]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1493953232.217489982]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1493953232.217580346]: ParallelPlan::solve(): Solution found by one or more threads in 0.002248 seconds
[ INFO] [1493953232.217779608]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1493953232.218106796]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1493953232.218333016]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1493953232.219015177]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1493953232.219192500]: ParallelPlan::solve(): Solution found by one or more threads in 0.001470 seconds
[ INFO] [1493953232.220524764]: SimpleSetup: Path simplification took 0.001177 seconds and changed from 4 to 25 states
```
```
[ INFO] [1493953234.182618807]: Received new trajectory execution service request...
[ INFO] [1493953234.182943515]: Returned 0 controllers in list
[ERROR] [1493953234.183068351]: Unable to identify any set of controllers that can actuate the specified joints: [ elbow_joint shoulder_lift_joint shoulder_pan_joint wrist_1_joint wrist_2_joint wrist_3_joint ]
[ERROR] [1493953234.183185327]: Known controllers and their joints:
```

**It seems no controllers to actuate the robot!**

gavanderhoorn, 2017-05-05:
If your UR5 (and control box) are a recent model (ie: a CB3 fi), then this is a know incompatibility between `ur_driver` and newer Polyscope versions.

Please see [wiki/universal_robot](http://wiki.ros.org/universal_robot) for the supported versions and our recommendation for users that have newer control boxes / Polyscope versions.

closed 2017-05-05 by gavanderhoorn

jacknlliu, 2017-05-05:
@gavanderhoorn  please update  `README.md` file of this repository for this  known incompatibility,and show the real tested versions about `ur_driver` and Polyscope. This will help more users who will contribute to this repository.

gavanderhoorn, 2017-05-05:
That is already documented on the ROS wiki:

> The `ur_driver` package has been successfully used with system versions ranging from `v1.5.7849` to `v1.8.23117`. Newer versions seem to have introduced various incompatible changes, and can at present not be used with the [ur_driver](http://wiki.ros.org/ur_driver) package.
> 
> We recommend users with newer system versions (`v3.x` and up) to use the [ur_modern_driver](https://github.com/ThomasTimm/ur_modern_driver) package. Robot description and [MoveIt](http://wiki.ros.org/MoveIt) configurations from the `universal_robot` stack can still be used. 

The ROS wiki is *the* place for package documentation and tutorials, so I'd rather not duplicate info like this into the readme. We could perhaps add an additional link from here to the ROS wiki pages to make this extra clear.

gavanderhoorn, 2017-05-05:
Just to reiterate: the ROS wiki has always been *the* place to document packages. Duplicating info in readmes was always discouraged. See some of the other `ros-industrial` repositories, you'll find they follow the same pattern.

We'll update the readme, but I just wanted to make this clear.

jacknlliu, 2017-05-05:
This repository doesn't follow the same pattern as other `ros-industrial` repositories. Other repositories show very clear link to their `ROS wiki`, but this repository does not clearly do this in its README.
Other `ros-industrial` repositories's pattern follow this:
> ROS-Industrial ABB meta-package. See the ROS wiki page for more information.
> The abb_experimental repository contains additional packages.

> ROS-Industrial Staubli meta-package. See the ROS wiki page for more information.

> ROS-Industrial Fanuc meta-package. See the ROS wiki page for more information.
> The fanuc_experimental repository contains additional packages.

Thank you very much for your reply. I think an additional link will be better.