ros-industrial/universal_robot #301: modify ur_driver/deserializeRT.py? (RobotStateRT, q_actual, unpacking RobotState Errors)

philipsgithub, 2017-05-31:
Hi universal-team,

I had the same issues like described in issue https://github.com/ros-industrial/universal_robot/issues/262. Could you please add support for RealTime_v3.4 like described in the xlsx-file at https://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/remote-control-via-tcpip-16496/? Check my latest answer on  issue https://github.com/ros-industrial/universal_robot/issues/262 for more details.

Missing support for the new version was causing these errors.

Best regard
Philip

gavanderhoorn, 2017-05-31:
>Missing support for the new version was causing these errors.

Thanks for reporting.

We're aware that `ur_driver` cannot be used with anything over 1.8.x, but we're currently not considering updating the deserialisers to support anything newer, as we'd like people to use the [ur_modern_driver](https://github.com/ThomasTimm/ur_modern_driver) with newer controllers.

I'm not saying that PRs that add support for the 3.x series of Polyscope won't be considered, but there are other aspect of `ur_driver` that made it worthwhile to migrate to `ur_modern_driver` completely.

See also the [wiki/universal_robot](http://wiki.ros.org/universal_robot) page, under *Compatibility*.

gavanderhoorn, 2017-09-13:
Closing as `ur_driver` will not be upgraded to be compatible with CB3 and up - it's deprecated. `ur_modern_driver` and the new Kinetic compatible rewrite should be used instead.

closed 2017-09-13 by gavanderhoorn