ros-industrial/universal_robot #304: UR10 does not move

nag92, 2017-07-20:
I am using ROS-Kinetic and ur_modern_drivers to connect to the arm.

I am able to connect to the ur10 arm and use the Moveit GUI to execute a trajectory. However after I execute a few trajectories the physical arm stops moving. The Moveit plug in for Rviz shows a legal trajectory when I press "plan" but when hit "execute" the terminal says that the execution was successful but the robot does not move. Is this a known issue? I cannot find anyone else who is having this problem.

gavanderhoorn, 2017-07-20:
This is probably not what you want to hear, but if you're using `ur_modern_driver` then it would be better if you'd open the issue over there.

Could I ask you to report it on the [ur_modern_driver/issues](https://github.com/ThomasTimm/ur_modern_driver/issues) tracker? Do first see whether ThomasTimm/ur_modern_driver#122 or ThomasTimm/ur_modern_driver#118 are related or describe your issue.

nag92, 2017-07-21:
It ended up being a problem with trying to send commands from the tablet and ROS at the same time.

closed 2017-07-21 by nag92

gavanderhoorn, 2017-07-21:
So can you give some more description of what was going on?

Were there multiple action clients preempting each other?