ros-industrial/universal_robot #306: The kinematics plugin (manipulator) failed to load.

lkj1114889770, 2017-07-26:
I want to use ik for UR3 provided by in the [https://github.com/ros-industrial/universal_robot/tree/kinetic-devel/ur_kinematics](url), and then i change the config for ur3 by modifying the kinematic.yaml  as shows in the follwing:

> manipulator:
  kinematics_solver: ur_kinematics/UR3KinematicsPlugin
  kinematics_solver_search_resolution: 0.005
  kinematics_solver_timeout: 0.005
  kinematics_solver_attempts: 3

however the probems occurs , it tells me that plugin ur_kinematics/UR3KinematicsPlugin.does not exit.

> [ERROR] [1501058251.852063276]: The kinematics plugin (manipulator) failed to load. Error: Could not find library corresponding to plugin ur_kinematics/UR3KinematicsPlugin. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1501058251.852186933]: Kinematics solver could not be instantiated for joint group manipulator

But it did have install the libur3_moveit_plugin.so in the folder ~/catkin_ws/devel/lib. 
How to solve the problem? Thanks in advance.

LuserBen, 2018-04-11:
Copy file  libur3_moveit_plugin.so( maybe and libur3_kin.so) in the folder ~/catkin_ws/devel/lib to the folder /universal_robot/ur_kinematics

gavanderhoorn, 2018-06-25:
This could be a configuration issue (ie: plugin not correctly built or present). Something to investigate perhaps.

gavanderhoorn, 2018-06-25:
I'm actually going to close this. Last comment by the OP was from July last year, and we need more information to reproduce and/or solve the problem.

@lkj1114889770: if this is still an issue for you, and you believe you have everything configured and setup correctly, then please re-open and provide us with more information.

closed 2018-06-25 by gavanderhoorn

2197808908a, 2024-04-10:
> This could be a configuration issue (ie: plugin not correctly built or present). Something to investigate perhaps.

What you said is right, it has reference significance