ros-industrial/universal_robot #317: UR5 Jerky Movement

animez4me, 2017-08-22:
Hi,

I am using moveit motion plan service and simply execute the path planned on a UR5. However, there are a lot of jerky movements, in severe cases, the robot's protective stop gets triggered. This occurs regardless the path was computed from getmotionplan or computecartesianpath. I have tried playing with the max_velocity and max_accelerations in the joints.yaml but not much luck. Even with acceleration at 0.05...

gavanderhoorn, 2017-09-13:
Just making sure: are you using `ur_driver` or `ur_modern_driver`? And what controller is this?

Also: perhaps related (but only if using `ur_modern_driver`): https://github.com/ThomasTimm/ur_modern_driver/issues/132.

gavanderhoorn, 2018-06-25:
Closing this due to no activity.

Again: not because I don't believe there is an actual issue, but because `ur_driver` should not be used and in the case it isn't being used other issue trackers are more appropriate.

If you feel this is not correct @wmy101, then please comment.

closed 2018-06-25 by gavanderhoorn