ros-industrial/universal_robot #324: RobotStateRT message with invalid package length is not handled properly

ipa-jfh, 2017-10-04:
This problem occured randomly (like every 10th time) with our:
- Real UR5 with CB2
- ROS Indigo 

Error:
After receiving the wrong package length of the RobotStateRT message the [`__run`](https://github.com/ros-industrial/universal_robot/blob/indigo-devel/ur_driver/src/ur_driver/driver.py#L357) function crashes. This leads to the problem that the driver does not read the joints states of the robot any longer. However, one can still send commands to the robot and then, as exeperienced, the robot will move uncontrolled.

From @ipa-mdl:
> Looks like a message [with invalid package length](https://github.com/ros-industrial/universal_robot/blob/indigo-devel/ur_driver/src/ur_driver/deserializeRT.py#L17) (or unaligned frame) was received and this is [not handled properly](https://github.com/ros-industrial/universal_robot/blob/indigo-devel/ur_driver/src/ur_driver/driver.py#L330).

Log of error:
```
[INFO] [WallTime: 1505466934.403690] No calibration offsets loaded from urdf
[INFO] [WallTime: 1505466934.405660] Max velocity accepted by ur_driver: 10.0 [rad/s]
[INFO] [WallTime: 1505466934.408669] Bounds for Payload: [0.0, 10.0]
Disconnected.  Reconnecting
[INFO] [WallTime: 1505466934.467400] Programming the robot
[INFO] [WallTime: 1505466934.470384] Programming the robot at 192.168.0.14
RobotStateRT has wrong length: 1254
Exception in thread URConnectionRT:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/nhg-hd/ur_training/src/universal_robot/ur_driver/src/ur_driver/driver.py", line 372, in __run
    self.__on_packet(packet)
  File "/home/nhg-hd/ur_training/src/universal_robot/ur_driver/src/ur_driver/driver.py", line 338, in __on_packet
    for i, q in enumerate(stateRT.q_actual):
AttributeError: 'RobotStateRT' object has no attribute 'q_actual'

Handling a request
[2017-09-15 11:15:35.018518] Out: hello
[INFO] [WallTime: 1505466935.035067] Robot connected
The action server for this driver has been started
[ERROR] [WallTime: 1505466935.619926] Stopped hearing from robot (last heard 1.108 sec ago).  Disconnected
Connection closed (command): 
Disconnected.  Reconnecting
[INFO] [WallTime: 1505466935.665214] Programming the robot
[INFO] [WallTime: 1505466935.670023] Programming the robot at 192.168.0.14
Handling a request
[2017-09-15 11:15:36.125733] Out: hello
[INFO] [WallTime: 1505466936.134888] Robot connected
[ERROR] [WallTime: 1505466936.326507] Stopped hearing from robot (last heard 1.815 sec ago).  Disconnected
Connection closed (command): 
Disconnected.  Reconnecting
```

gavanderhoorn, 2018-06-25:
`ur_driver` is deprecated and will not see any new development, but I'll leave it up to you @ipa-jfh whether this is something you want to spend time on and fix.

I've tagged it with `wrid18` as it could be something to do that day, but I'll already assign it to you so it won't be picked up by any other participant.

If this isn't an issue with any of the replacements you might want to consider using those.

Checking package sizes before deserialising them shouldn't be too hard to add though, so it might be a nice issue to get into the code.

chandanlal92, 2019-04-02:
same issue ur10 cb3

chandanlal92, 2019-04-02:
ros kinetic

miguelprada, 2019-04-02:
Which PolyScope version are you running? The driver in this repo has been long deprecated and I don't think it supports PolyScope versions 3.x. Please try with the [ur_modern_driver](https://github.com/ros-industrial/ur_modern_driver/tree/kinetic-devel) (make sure to use `kinetic-devel` branch).

chandanlal92, 2019-04-02:
Ya it worked I changed ur_modern_driver

ipa-hsd, 2019-06-19:
Shouldn't this package be removed completely if it is deprecated?

gavanderhoorn, 2019-11-23:
I'm closing this as with the deprecation of `ur_modern_driver` we'll also remove `ur_driver` from this repository in `melodic-devel` (#448).

Refer to the [announcement on ROS Discourse](https://discourse.ros.org/t/deprecation-of-ur-modern-driver/11629).

closed 2019-11-23 by gavanderhoorn