ros-industrial/universal_robot #327: Add effort and velocity hardware interface joint trajectory controller

jacknlliu, 2017-11-05:
Could add effort and velocity hardware interface joint trajectory controller? Currently, only position joint controller in this repo.

gavanderhoorn, 2017-11-05:
This would probably be more appropriate in `ur_modern_driver`, but without support in the controller for `effort`, how would that work exactly?

`ur_modern_driver` already has a velocity `ros_control` configuration.

jacknlliu, 2017-11-05:
I think ros controllers for gazebo simulator also can be configured. `ur_modern_driver` use `velocity` interface directly on the real robot, we also need the simulator, isn't it? And I also think `effort` interface can be configured for research.

gavanderhoorn, 2017-11-05:
> I think ros controllers for gazebo simulator also can be configured. 

Ah, it was not clear you were talking about simulation.

> we also need the simulator, isn't it? And I also think `effort` interface can be configured for research.

Well, yes, `gazebo_ros_control` could support `effort`, but I'm always wondering how useful that is, seeing as the real hardware doesn't support it.

jacknlliu, 2017-11-05:
OK. I think you want to make this repo more conservative, at least `velocity` interface joint trajectory controller can be configured in this repo?

gavanderhoorn, 2017-11-05:
Would #227 not already do what you want?

jacknlliu, 2017-11-05:
No. #227 don't ship with joint trajectory controller.

gavanderhoorn, 2017-11-05:
Well no, I guess that is left as an exercise 'for the reader'. But it's really basically just a copy-paste of what is already in the `ur_gazebo/controller` dirs.

The xacro extensions implement most of the needed infrastructure though.

jacknlliu, 2017-11-05:
:smile: OK. So nothing will be done just leave them for future?

gavanderhoorn, 2017-11-05:
We can certainly merge a PR against `ur_gazebo`, but that would need #227 to be merged as well though.

If you submit a PR that adds the necessary `ros_control` config yamls that would get reviewed. I'm not in a position to add the necessary configuration myself though.

jacknlliu, 2017-11-05:
Actually, I have made these in local research, but some parameters still need to be more tuned. And I can submit a basic PR for support these interface including URDF and gazebo controller configs. But these need the community to test and improve it. I think this is better for the community who use this repo to do research on robotics, so I file this ticket as marked.

gavanderhoorn, 2017-11-05:
I'm not entirely sure I understand you: you already have the required changes, but are not confident they are 'good enough'? Would submitting a PR not allow other users to check what you did and then we can decide collectively whether it would be a good idea to merge the PR?

jacknlliu, 2017-11-05:
I already have done in my custom package. But I need more test, I think these may be good for this upstream. I will submit a PR for more test about it.

gavanderhoorn, 2018-06-25:
Closing due to no activity.

If this is still something you would be interested in, please consider submitting a PR.

closed 2018-06-25 by gavanderhoorn