ros-industrial/universal_robot #328: UR3 hardware cannot move with Moveit Python interface

Qifan94, 2017-11-20:
Hi,
I tested the ur_modern_driver with the hardware. The test_move.py in ur_modern_driver pkg did work and the plan and execute function in Rivz also could work with the hardware. But my code cannot set a target and move it to the target. I tried my code in a simulation in Rivz with UR3 robot fake controller. But When I change to control the real robot it does not work anymore and the error is shown as following.

[ERROR]: Unable to construct goal representation
[INFO]: ABORTED: Catastrophic failure

My python code is here.

`#!/usr/bin/env python
##
## To use the python interface to move_group, import the moveit_commander
## module.  We also import rospy and some messages that we will use.
import sys
import copy
import rospy
import moveit_commander
import moveit_msgs.msg
import geometry_msgs.msg
import numpy as np
import scipy.io as sio
## END_SUB_TUTORIAL

from std_msgs.msg import String

def move_group_python_tutorial():
  ## BEGIN_TUTORIAL
  ##
  ## Setup
  ## ^^^^^
  ## CALL_SUB_TUTORIAL imports
  ##
  ## First initialize moveit_commander and rospy.
  print "============ Starting tutorial setup"
  moveit_commander.roscpp_initialize(sys.argv)
  rospy.init_node('move_group_python_tutorial',
                  anonymous=True)

  ## Instantiate a RobotCommander object.  This object is an interface to
  ## the robot as a whole.
  robot = moveit_commander.RobotCommander()

  ## Instantiate a PlanningSceneInterface object.  This object is an interface
  ## to the world surrounding the robot.
  scene = moveit_commander.PlanningSceneInterface()

  ## Instantiate a MoveGroupCommander object.  This object is an interface
  ## to one group of joints.  In this case the group is the joints in the left
  ## arm.  This interface can be used to plan and execute motions on the left
  ## arm.
  group = moveit_commander.MoveGroupCommander("arm")

  ## We create this DisplayTrajectory publisher which is used below to publish
  ## trajectories for RVIZ to visualize.
  display_trajectory_publisher = rospy.Publisher(
                                      '/move_group/display_planned_path',
                                      moveit_msgs.msg.DisplayTrajectory)

  ## Wait for RVIZ to initialize. This sleep is ONLY to allow Rviz to come up.
  print "============ Waiting for RVIZ..."
  rospy.sleep(3)
  print "============ Starting tutorial "

  ## Getting Basic Information
  ## ^^^^^^^^^^^^^^^^^^^^^^^^^
  ##
  ## We can get the name of the reference frame for this robot
  print "============ Reference frame: %s" % group.get_planning_frame()

  ## We can also print the name of the end-effector link for this group
  print "============ Reference frame: %s" % group.get_end_effector_link()

  ## We can get a list of all the groups in the robot
  print "============ Robot Groups:"
  print robot.get_group_names()

  ## Sometimes for debugging it is useful to print the entire state of the
  ## robot.
  print "============ Printing robot state"
  print robot.get_current_state()
  print "============"

  ## Planning to a Pose goal
  ## ^^^^^^^^^^^^^^^^^^^^^^^
  ## We can plan a motion for this group to a desired pose for the 
  ## end-effector
  

 
  ## You can ask RVIZ to visualize a plan (aka trajectory) for you.  But the
  ## group.plan() method does this automatically so this is not that useful
  ## here (it just displays the same trajectory again).

  ## Moving to a pose goal
  ## ^^^^^^^^^^^^^^^^^^^^^
  ##
  ## Moving to a pose goal is similar to the step above
  ## except we now use the go() function. Note that
  ## the pose goal we had set earlier is still active 
  ## and so the robot will try to move to that goal. We will
  ## not use that function in this tutorial since it is 
  ## a blocking function and requires a controller to be active
  ## and report success on execution of a trajectory.
## Planning to a Pose goal
  ## ^^^^^^^^^^^^^^^^^^^^^^^
  ## We can plan a motion for this group to a desired pose for the 
  ## end-effector
  pose_ready = group.get_current_pose().pose
  print pose_ready
  #group.set_pose_target(pose_ready)
  #group.plan()
  #rospy.sleep(1)
  #group.go(wait=True)
  print 'Move to Ready Position'
  print "============ Generating plan 1"
  
  n = 36
  R = 0.1
  i = 0
  sio.savemat('test_data_python.mat',{ 'a': 0, 'b': 9, 'c':18, 'd' : 27 })
  data = sio.loadmat('test_data_python.mat')
  print data
  x1 = int(data['a'])
  x2 = int(data['b'])
  x3 = int(data['c'])
  x4 = int(data['d'])
  line = [x1,x2,x3,x4]
  while True:
	ang = line[i] *2 * np.pi / n
 
  	#print geometry_msgs.msg.Pose(
  	#print group.get_current_pose().pose
  	pose_target = group.get_current_pose().pose
  	print "=============Print pose_target now:"
        print line[i]
  	print pose_target
  	pose_target.orientation.w += 0.0
  	pose_target.position.x -= R*np.sin(ang)
  	pose_target.position.y -= 0
  	pose_target.position.z += R*np.cos(ang)
 	print pose_target
  	group.set_pose_target(pose_target)

	group.plan()
 	rospy.sleep(3)
  	group.go(wait=True)
	rospy.sleep(3)
	i = i+1
			
  ## Now, we call the planner to compute the plan
  ## and visualize it if successful
  ## Note that we are just planning, not asking move_group 
  ## to actually move the robot
  	
  

  	print "============ Waiting while RVIZ displays plan1..."
  	rospy.sleep(5)
	group.set_pose_target(pose_ready)
  	group.plan()
	rospy.sleep(3)
  	group.go(wait=True)
  	rospy.sleep(3)
        if i == 4:
		break
	
  # Uncomment below line when working with a real robot
  # group.go(wait=True)

 
  ## Adding/Removing Objects and Attaching/Detaching Objects
  ## ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  ## First, we will define the collision object message
  collision_object = moveit_msgs.msg.CollisionObject()

  ## When finished shut down moveit_commander.
  moveit_commander.roscpp_shutdown()

  ## END_TUTORIAL

  print "============ STOPPING"

if __name__=='__main__':
  try:
    move_group_python_tutorial()
  except rospy.ROSInterruptException:
    pass
`

gavanderhoorn, 2017-11-24:
If this is still a problem, please re-open over at [ur_modern_driver/issues](https://github.com/ThomasTimm/ur_modern_driver/issues), seeing as you're trying to use `ur_modern_driver`.

gavanderhoorn, 2017-11-24:
> `pose_target.orientation.w += 0.0`

are you sure this is ok?

`orientation` is a [quaternion](http://wiki.ros.org/Tutorials/Quaternions), not an RPY triple.

gavanderhoorn, 2017-11-24:
I'm closing this as in any case this is not an issue with the pkgs in this repository.

closed 2017-11-24 by gavanderhoorn