ros-industrial/universal_robot #330: Moveit control UR3 real (3.40). Runtime error , the program uses too much time without directing the robot ..wait or sync () to split the robot's long program sequence without moving it

naijoug, 2017-11-24:
I use moveit control UR3 real. When planning distance is far, the robot will stop. PolyScope show the error info : " Runtime error, the program uses too much time without directing the robot ..wait or sync () to split the robot's long program sequence without moving it".

gavanderhoorn, 2017-11-24:
Are you using `ur_driver`, or `ur_modern_driver`?

If the first: that is completely unsupported on CB3 controllers.

If the second: please post this issue over at [ur_modern_driver/issues](https://github.com/ThomasTimm/ur_modern_driver/issues).

naijoug, 2017-11-24:
@gavanderhoorn Thank you.  I use `ur_modern_driver`.  I have already post it.

gavanderhoorn, 2017-11-24:
In that case that was a cross-post which isn't very nice, but ok.

closed 2017-11-24 by gavanderhoorn