ros-industrial/universal_robot #333: Next Kinetic Release

davetcoleman, 2017-12-17:
I submitted a PR a few months ago that has not been released to the debian packages yet, can a maintainer please make a new ROS buildfarm release?

ipa-fxm, 2018-01-05:
I will trigger new releases for care-o-bot over the weekend anyway, so I don't mind doing it for one more repo - if my permissions are still sufficient :wink:

I will release both indigo and kinetic and merge the following PRs - if I do not get any objections:
 - #312 
 - #322 
 - #329 
 - #332 

@gavanderhoorn @davetcoleman @Levi-Armstrong @shaun-edwards

ipa-fxm, 2018-01-06:
I'm trying to release, but https://github.com/ros/catkin/pull/902 keeps messing around which hasn't been released yet...

ipa-fxm, 2018-01-06:
There you go:
 - Indigo: https://github.com/ros/rosdistro/pull/16707
 - Kinetic:

closed 2018-01-06 by ipa-fxm