ros-industrial/universal_robot #339: Invoking "make VERBOSE=1 -j4 -l4" failed

SantaHub, 2018-02-12:
[ 92%] Linking CXX shared library /home/austin/Robo/catkin_ws/devel/lib/libur10_moveit_plugin.so
cd /home/austin/Robo/catkin_ws/build/universal_robot/ur_kinematics && /usr/bin/cmake -E cmake_link_script CMakeFiles/ur10_moveit_plugin.dir/link.txt --verbose=1
/usr/bin/c++  -fPIC   -shared -Wl,-soname,libur10_moveit_plugin.so -o /home/austin/Robo/catkin_ws/devel/lib/libur10_moveit_plugin.so ......

[ 92%] Built target ur10_moveit_plugin
Makefile:138: recipe for target 'all' failed

Looks like c++ error ?? How to resolve?

gavanderhoorn, 2018-02-12:
I don't see an actual error message - just the overal `make` exit status - and that recipe `all` failed.

Just did a quick build in a clean Docker container and everything compiled and linked.

Without more information - and the actual error - there is not much I can do to help.

SantaHub, 2018-02-12:
I took a new terminal and run the code. 
```
echo $ROS_PACKAGE_PATH
/opt/ros/kinetic/share
```
Now its success,

closed 2018-02-12 by SantaHub