ros-industrial/universal_robot #340: UR3, 5, 10 dual robot arm

Purdiding, 2018-02-12:
HI all,

I want to make a dual robot arm using two ur3.

Therefore, I try to revise the urdf(xacro) of UR3.
However, I don't know how to make a left arm, right arm part.

How to start this work?
There is any reference web site or something?

Thank you.

miguelprada, 2018-02-13:
See how [`ur3_robot.urdf.xacro`](https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/ur_description/urdf/ur3_robot.urdf.xacro) instantiates a single UR3 using the macro defined in [`ur3.urdf.xacro`](https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/ur_description/urdf/ur3.urdf.xacro).

You should do something similar except instancing the macro twice, and using the `prefix` argument to prepend something like `left_` and `right_` to each instance respectively.

You can then use a launchfile similar to [`ur3_upload.launch`](https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/ur_description/launch/ur3_upload.launch) to process your model with [`xacro`](http://wiki.ros.org/xacro) and upload it to the parameter server.

gavanderhoorn, 2018-02-13:
@Purdiding: just a note: what you're trying to do is not specific to the packages in this repository. I would recommend to broaden your search to working with xacro and how to combine multiple xacro macros into a single top-level xacro.

@miguelprada has given you some pointers as to which files you should use for this.

gavanderhoorn, 2018-02-13:
As this is not really an issue / problem with the pkgs in this repository, I'm closing this question.

@Purdiding: if you have any further questions about how to work with xacro, I would recommend to consider posting questions on `answers.ros.org`.

gavanderhoorn, 2018-02-13:
Note that you may run into #247.

closed 2018-02-13 by gavanderhoorn