ros-industrial/universal_robot #341: IKFast vs ur_kinematics

mlautman, 2018-02-13:
I have been working on porting some features available in the ur_kinematics into the ikfast template so that all robots can benefit from the enhancements @kphawkins and the other contributors built into the ur_kinematics solver. It seems pretty clear that ur_moveit_plugin.cpp was heavily influenced by the ikfast generated templates. My question is beyond the obvious, what would be involved in porting the enhancements back into the ikfast template? I'm particularly interested in supporting the ability to use the robot arm as a subset of a kinematic chain but additional features like setRedundantJoints are interesting as well.

gavanderhoorn, 2018-02-14:
> I'm particularly interested in supporting the ability to use the robot arm as a subset of a kinematic chain

It's interesting that you post this, as #247 is a heated discussion about doing exactly the opposite.

> additional features like setRedundantJoints are interesting as well.

This confuses me, as that is already part of the IKFast plugin wrapper (see [here](https://github.com/ros-planning/moveit/blob/94b59aa1c191c4896aaefde9cfb9b0dee3f60176/moveit_kinematics/ikfast_kinematics_plugin/templates/ikfast61_moveit_plugin_template.cpp#L284-L298)).

gavanderhoorn, 2018-02-14:
To structure this a bit: can you point to some of the things that you see missing in the IKFast wrappers that `ur_kinematics` has?

mlautman, 2018-02-17:
The ur_kinematics package is able to recognize if there are additional links in the group that aren't part of the ur chain and transforms input poses from the world frame to the appropriate frame for the analytic solution. The relevant code for these transformations [here.](https://github.com/ros-industrial/universal_robot/blob/ad740e0dd028cd9614b869d306dfc2da402cbc46/ur_kinematics/src/ur_moveit_plugin.cpp#L617-L647) I realize that this same functionality can be achieved outside of the kinematic solver quite easily but the use case we have in mind involves putting the transformations intot the kinematic solver itself.