ros-industrial/universal_robot #347: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'

Mylittledream, 2018-03-16:
Hi,
   When I use the "ur5_moveit_cofig",the terminal display
`[ WARN] [1521188902.220001947]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'`
  Does this affect the real UR robot?
  Please help to solve this

gavanderhoorn, 2018-03-23:
This is a known issue, see ros-industrial/universal_robot#284. See the discussion at the end for a potential work-around.

closed 2018-03-23 by gavanderhoorn

matteolucchi, 2018-10-22:
Hoping that this will be helpful for other people with this issue I post here the two additional urdf created following the suggestion from @jacknlliu and @gavanderhoorn  . I post here the ones created for UR10 

`ur10_moveit.urdf.xacro` to be added in /ur_description/urdf
```
<?xml version="1.0"?>
<robot xmlns:xacro="http://ros.org/wiki/xacro"
       name="ur10" >

  <!-- common stuff -->
  <xacro:include filename="$(find ur_description)/urdf/common.gazebo.xacro" />

  <!-- ur10 -->
  <xacro:include filename="$(find ur_description)/urdf/ur10.urdf.xacro" />

  <!-- arm -->
  <xacro:ur10_robot prefix="" joint_limited="false"/>

</robot>
```
`ur10_joint_limited_moveit.urdf.xacro` to be added in /ur_description/urdf
```
<?xml version="1.0"?>
<robot xmlns:xacro="http://ros.org/wiki/xacro"
       name="ur10" >

  <!-- common stuff -->
  <xacro:include filename="$(find ur_description)/urdf/common.gazebo.xacro" />

  <!-- ur10 -->
  <xacro:include filename="$(find ur_description)/urdf/ur10.urdf.xacro" />

  <!-- arm -->
  <xacro:ur10_robot prefix="" joint_limited="true"/>

</robot>
```

edit  ur10_moveit_config/launch/planning_context.launch referencing the 2 new file just added
```
<!-- Load universal robot description format (URDF) -->
  <group if="$(arg load_robot_description)">
    <param unless="$(arg limited)" name="$(arg robot_description)" command="$(find xacro)/xacro --inorder '$(find ur_description)/urdf/ur10_moveit.urdf.xacro'" />
    <param if="$(arg limited)" name="$(arg robot_description)" command="$(find xacro)/xacro --inorder '$(find ur_description)/urdf/ur10_joint_limited_moveit.urdf.xacro'" />
  </group>
```

edit  ur10_moveit_config/launch/move_group.launch adding the arg to load the robot_description
```
  <include file="$(find ur10_moveit_config)/launch/planning_context.launch">
    <arg name="limited" value="$(arg limited)" />
    <arg name="load_robot_description" value="true" />
  </include>
```

puqian13, 2019-09-17:
我有同样的问题，请问怎么解决

gavanderhoorn, 2019-09-17:
@puqian13: please do not post comments in languages other than English.

puqian13, 2019-09-17:
Ｉhave a same question.How did you solve it

byteofsoren, 2020-05-06:
I to have the same problem. 
System Jetson Xavier Ubuntu 18.04
Ros version Melodic

Our worskpace_ws/src/
Contains 
```
src/
   CMakeLists.txt
   universal_robot/
   robot_py/
```
The code we are trying to run is:
```
#!/usr/bin/env python
import rospy, sys
import moveit_commander
from math import pi
rom geometry_msgs.msg import Pose
from copy import deepcopy

class MoveItCartesianPath:
    def __init__(self):
        rospy.init_node("moveit_cartesian_path", anonymous=False)

        rospy.loginfo("Starting node moveit_cartesian_path")

        rospy.on_shutdown(self.cleanup)

        # Initialize the move_group API
        moveit_commander.roscpp_initialize(sys.argv)

        # Initialize the move group for the ur5_arm
        self.arm = moveit_commander.MoveGroupCommander('manipulator')

        # Get the name of the end-effector link
        end_effector_link = self.arm.get_end_effector_link()

        # Set the reference frame for pose targets
        reference_frame = "base_link"

        # Set the ur5_arm reference frame accordingly
        self.arm.set_pose_reference_frame(reference_frame)

        # Allow replanning to increase the odds of a solution
        self.arm.allow_replanning(True)

        # Allow some leeway in position (meters) and orientation (radians)
        self.arm.set_goal_position_tolerance(0.01)
        self.arm.set_goal_orientation_tolerance(0.1)

        # Move the robot to a better position.
       	joint_goal = self.arm.get_current_joint_values()
        joint_goal[0] = 0
        joint_goal[1] = -pi/2
        joint_goal[2] = pi/4
        joint_goal[3] = -pi/2 - pi/4
        joint_goal[4] = -pi/2
        joint_goal[5] = pi/3
        #self.joint_goal[6] = 0
        print "joint_goal = %s" % str(joint_goal)
        self.arm.go(joint_goal, wait=True)

        # Get the current pose so we can add it as a waypoint
        self.start_pose = self.arm.get_current_pose(end_effector_link).pose
	# Mycode

        # Initialize the waypoints list
        # waypoints = []

        # # Set the first waypoint to be the starting pose
        # # Append the pose to the waypoints list
        # waypoints.append(start_pose)

        # wpose = deepcopy(start_pose)

        # # Set the next waypoint to the right 0.5 meters
        # wpose.position.x -= -1.1

        # waypoints.append(deepcopy(wpose))

        fraction = 0.0
        maxtries = 100
        attempts = 0

        # Set the internal state to the current state
        self.arm.set_start_state_to_current_state()

        # Plan the Cartesian path connecting the waypoints
        while fraction < 1.0 and attempts < maxtries:
	    self.plan_x(0.1)  #Go 10cm in x directoion.
            (plan, fraction) = self.arm.compute_cartesian_path (self.waypoints, 0.01 , 0.0, True)
	    # rospy.loginfo("plan =\n{}".format(plan))
            # Increment the number of attempts
            attempts += 1

            # Print out a progress message
            if attempts % 10 == 0:
                rospy.loginfo("Still trying after " + str(attempts) + " attempts...")

        # If we have a complete plan, execute the trajectory
        if fraction == 1.0:
            rospy.loginfo("Path computed successfully. Moving the arm.")
            ret = False
            attempts = 0
            while attempts < maxtries or ret:
                ret = self.arm.execute(plan)
                # rospy.loginfo("attempts={}, \tret={}".format(attempts, "Works" if ret else "Did not work"))
                attempts += 1
            rospy.loginfo("Path execution complete.")
        else:
            rospy.loginfo("Path planning failed with only " + str(fraction) + " success after " + str(maxtries) + " attempts.")

    def plan_x(self,x):
        self.waypoints = []
        # Set the first waypoint to be the starting pose
        # Append the pose to the waypoints list
        self.waypoints.append(self.start_pose)
        wpose = deepcopy(self.start_pose)
        # Set the next waypoint to the right 0.5 meters
        wpose.position.x += x
        wpose.position.z += -0.1

        self.waypoints.append(deepcopy(wpose))

    def cleanup(self):
       	joint_goal = self.arm.get_current_joint_values()
        # Reset the robot position.
        joint_goal[0] = 0
        joint_goal[1] = 0
        joint_goal[2] = 0
        joint_goal[3] = 0
        joint_goal[4] = 0
        joint_goal[5] = 0
        #self.joint_goal[6] = 0
        self.arm.go(joint_goal, wait=True)
        rospy.loginfo("Stopping the robot")

        # Stop any current arm movement
        self.arm.stop()

        #Shut down MoveIt! cleanly
        rospy.loginfo("Shutting down Moveit!")
        moveit_commander.roscpp_shutdown()
        moveit_commander.os._exit(0)

 if __name__ == "__main__":
    try:
        r=MoveItCartesianPath()
        r.cleanup()
    except KeyboardInterrupt:
        print "Shutting down MoveItCartesianPath node."
```
But at `MoveGroupCommander` I get the following error. 
```
[ INFO] [1588771909.232586001]: Loading robot model 'ur5'...
[ WARN] [1588771909.236208271]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1588771909.236379224]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1588771909.426999340, 3657.502000000]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/robot_description_kinematics/manipulator/kinematics_solver_attempts' from your configuration.
[ INFO] [1588771910.698520914, 3658.668000000]: Ready to take commands for planning group manipulator.
```
Then I assume because of the skipped link we get the error:
```
[ INFO] [1588771917.904498161, 3665.076000000]: ABORTED: Solution found but controller failed during execution
```
Does anyone know how to fix this?

gy2256, 2021-04-02:
I have this problem as well and it is driving me crazy. Any help would be appreciated!