ros-industrial/universal_robot #348: How to run ur_gazebo without visual simulation?

pipothy, 2018-03-26:
I'm running ur_gazebo for the ur5 on a virtualbox ubuntu installation and as one can probably guess, it's painfully slow. The simulator barely gets going and the frame rate is awful. When I try boot up RViz as well, it doesn't even get past initialising.

Is there any way I can run the simulator without the visualisation of the robot? I assume that's the bottleneck, or will this just not be possible on a virtualbox?

gavanderhoorn, 2018-03-26:
Just set the `gui` launch file argument to `false` ([this one](https://github.com/ros-industrial/universal_robot/blob/ad740e0dd028cd9614b869d306dfc2da402cbc46/ur_gazebo/launch/ur10.launch#L5) for the UR 10).

gavanderhoorn, 2018-03-26:
I'm going to close this, as it's not an issue with the packages in this repository.

Feel free to keep commenting though.

closed 2018-03-26 by gavanderhoorn