ros-industrial/universal_robot #351: Rviz UR5 not working as expected

AndrewJSchoen, 2018-04-28:
On the [ROS page for kinetic](http://wiki.ros.org/ur_description?distro=kinetic), it says the following:

To view and manipulate the arm models in rviz, install the package from package management and launch the following:
```
roslaunch ur_description ur5_upload.launch
roslaunch ur_description test.launch
```

However, when I run the first command (`ur5_upload.launch`), it says 
```No processes to monitor
shutting down processing monitor...
... shutting down processing monitor complete.
```
When I run the second, it brings up rviz, but no robot arm, as well as an error that the `Fixed Frame [map] does not exist`
I have installed the universal robots package using apt get:
```sudo apt-get install rod-kinetic-universal-robot```

gavanderhoorn, 2018-04-29:
I've just tested this (in a Docker container, so clean system) and for me, `roslaunch ur_description ur5_upload.launch` gives me the following output:

```
root@f47bf317a8f7:/# roslaunch ur_description ur5_upload.launch 
... logging to /root/.ros/log/9bb23a04-4b89-11e8-aa75-0242ac110002/roslaunch-f47bf317a8f7-488.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://f47bf317a8f7:35431/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES

auto-starting new master
process[master]: started with pid [501]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 9bb23a04-4b89-11e8-aa75-0242ac110002
process[rosout-1]: started with pid [514]
started core service [/rosout]
^C
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Note how I have to explicitly kill/stop everything with `ctrl+c`. If `roslaunch` exits immediately for you, something would seem to be not working correctly.

One thing you could be running into is #313. If you really only installed `ros-kinetic-universal-robot`, it could be that you're missing `xacro`, but I would've expected you to have noticed and reported that (as it results in a Python backtrace of about 30 lines being printed on the console).

If you're not missing `xacro`, then please provide us with more info (ie: more of the output of `roslaunch`). Including as much of the relevant output as possible is always a good idea, as without this, we cannot help you.

gavanderhoorn, 2018-05-01:
I'm getting the feeling something is not correctly configured on your machine, or something is interfering with the normal `roslaunch` control flow.

I'm also starting to doubt whether this is something caused by the packages in this repository.

What happens when you just run `roscore` by itself?

Do you happen to have a `roscore` running already before you start `roslaunch`?

Notice how in your `roslaunch` output we don't see `roslaunch` starting `rosout`, nor setting the `runid` parameter. Do you have `rosout` installed (what is the output of `dpkg -l | grep rosout`)? And to make sure: how did you install ROS? How did you setup your workspace? What made you add `xacro` to your workspace instead of installing it using `apt-get install ros-kinetic-xacro`?

gavanderhoorn, 2018-05-09:
@AndrewJSchoen: is this still an issue for you?

atyshka, 2018-05-23:
@gavanderhoorn I have been having this issue. I'm using ur modern driver with the moveit config packages for a ur10 and I'm getting exactly the same issue as Andrew. Wasn't a problem a month ago and I've changed nothing.

gavanderhoorn, 2018-05-24:
The packages haven't changed either, so I'm inclined to believe this is a configuration change on your end.

Have you set `ROS_IP`, `ROS_HOSTNAME` or any of the other networking related environment variables in your `$HOME/.bashrc` (for instance)? Switched networking connections?

>  I'm using ur modern driver with the moveit config packages for a ur10

I'm not sure you're having the exact same issue then: @AndrewJSchoen wasn't using a driver, he was just testing the urdf.

gavanderhoorn, 2018-06-25:
I'm going to close this due to no activity.

@AndrewJSchoen and @atyshka: if this is still an issue we would need more information.

Please feel free to re-open but do provide more details and / or steps to reproduce this.

closed 2018-06-25 by gavanderhoorn

ChunJyeBehBeh, 2019-03-01:
> When I run the second, it brings up rviz, but no robot arm, as well as an error that the `Fixed Frame [map] does not exist`

I have encountered the same problem, I change the Fixed Frame under Global Options, then the status of Fixed  Frame become OK but the no robot arm show up.

May I know how to solve it?

atyshka, 2019-03-01:
You forgot the config=true argument

ChunJyeBehBeh, 2019-03-01:
@atyshka  May I know config=true argument at where?

atyshka, 2019-03-01:
For the launch file