ros-industrial/universal_robot #352: UR5 motion velocity cannot be adjusted using move_group API

luhongliang, 2018-05-09:
Hello! I do some tracking object task using UR5. The trajectory is planned and executed through move_group and is controlled via ur_modern_driver ros package you provided. I want UR5 to move as slowly as it can, so I reduce max_velocity and max_acceleration and reduce the parameter value of planning step in computeCartesianPath function provided by move_group.h header. Robot is running well. When I want it to move more slowly, I try to set max velocity to 0,02 in source file using setMaxVelocityScalingFactor API, but robot motion velocity does not change at all. I don't know why and hope you can provide some help, thanks!

gavanderhoorn, 2018-05-09:
This would seem to be a duplicate of ThomasTimm/ur_modern_driver#172: as I wrote there, the limitation is in MoveIt, not in the driver, or the packages in this repository.

See [MoveIt velocity scaling for Cartesian Path](https://answers.ros.org/question/288989) on ROS Answers for some more info.

gavanderhoorn, 2018-05-09:
As this is also not an issue with this repository, I'm going to close this.

closed 2018-05-09 by gavanderhoorn