ros-industrial/universal_robot #353: Protective stop message during motion

rajmohan747, 2018-05-28:
Hi, I am using UR5 and after doing the trajectory planning during some point I am getting protective stop message in teach pendant.It happens at times when a jerky kind of motion happens.

Can some one tell how to tackle this situation.
What should be the parameter to be changed for obtaining a smoother motion.?

If max_acceleration: and max_velocity in **_joint_limits.yaml_** will do this,what would be a good range of values to be supplied.?

philip-long, 2018-05-28:
Hi, I was getting a similar error because of inertial forces when operating in a reduced mode. Some things you could try

- If you're using a tool make sure the inertial parameters are entered correctly. 
- The  jerky motion may be caused by a bad connection between the robot and external PC
- If you're using moveit to generate trajectories try to increase the time between waypoints by reducing maximum velocity in joint_limits

rajmohan747, 2018-05-29:
Hi @philip-long  
Thanks for your quick response
I am not using any tool as of now.
I tried reducing the velocity values to 75 % of default one in the joint_limits.yaml

Also ,earlier the kinematics_solver was kdl_kinematics__plugin/KDLKinematicsPlugin_ , and now I switched to trac_ik_kinematics_**_plugin/TRAC_IKKinematicsPlugin_**.

It makes the motion somewhat better .But still facing the jerky motion and protective stop issues at some random instance. Motion is being executed through same way points only.Some times it works smoothly and some random occasions its throwing issue)=

gavanderhoorn, 2018-05-29:
First: please make sure you're not using a wireless connection.

If you're not and problems persist, see whether [this fork](https://github.com/Zagitta/ur_modern_driver) works better. You might even try [this PR](https://github.com/Zagitta/ur_modern_driver/pull/9) against that fork.

In general jerky motion is caused -- as @philip-long mentions -- by network limitations and or CPU usage of the PC controlling the motion.

rajmohan747, 2018-05-31:
@gavanderhoorn I am using wired connection only.Also I don't think its about the CPU usage of PC created issue as its having good specifications and not other process running in background.

I am not facing the issues for last couple of days.Yet to confirm whether its stable or not.

Here are the changes that were made.

In **kinematics.yaml** ,used ,kinematics_solver: trac_ik_kinematics_plugin/TRAC_IKKinematicsPlugin.
Reduced the max_velocity and max_acceleration to 50% of default values in **joint_limits.yaml**
Added default_robot_padding: 0.07 in **padding.yaml.**

gavanderhoorn, 2018-05-31:
Reducing max velocities and accelerations is of course only a work-around.

Btw: which driver are you using? If not `ur_driver`, then this issue should be closed and you should report it on the appropriate issue tracker.

gavanderhoorn, 2018-06-25:
I'm going to close this. Not because I don't believe you have a problem @rajmohan747 with the motion, but because if this is with `ur_driver`, it should not be used, and if this isn't with `ur_driver`, the issue should be opened on a more appropriate issue tracker.

If you don't agree, please comment on this issue again.

closed 2018-06-25 by gavanderhoorn