ros-industrial/universal_robot #354: Joint velocities in gazebo are all near to zero

geweihgg, 2018-05-29:
Hi, 
I'm using ur5 in gazebo, and I want to get the joint velocities. So I let the arm moves all the time, and then run the following command line:
```
rostopic echo /joint_states
```
But the joint velocities in the output are like this:
[image]
Um... It seems that these joint velocities infomation are all noises, how can I fix it?
Looking forward to your help.
Thank you very much.
GeWei

gavanderhoorn, 2018-05-29:
A first step would be to reproduce this with just the UR in Gazebo - remove the gripper.

If that still shows unexpected joint velocities, then next would be to check the `gazebo_ros_control` plugins your using. We're not doing anything special here in this repository, we're only just users of those.

---

In any case: you're reporting a problem with something, but don't provide any information on which version of the involved components you are using. So at the very least, include the version of ROS, how you installed it, version of Gazebo, how you installed that, version of `gazebo_ros_control`, and how you installed that, and provide some more detail on how you are interfacing with all of this.

geweihgg, 2018-05-29:
Hi, @gavanderhoorn ,
Thank you very much, I'll remove the gripper and check if it works.
If the problem still exists, I'll ask for advice. And thank you for you suggestion about how to reporting a problem.
Thank you~
GeWei

geweihgg, 2018-05-30:
Hi, @gavanderhoorn 
Sorry to interrupt you again. I've tested the ur10 gazebo model, there being no valid velocities too. And when I test the baxter robot in gazebo, velocities info from /joint_states topic seems ok. The testing environments  are as follows:
```
ROS indigo
Gazebo 2.2.6
```
It seems that ```gazebo_ros_control``` is ok on my computer, since that velocities testing with baxter is ok.
Now I runs the following commands:
```
roslaunch ur_gazebo ur10_joint_limited.launch
roslaunch ur10_moveit_config demo.launch
rostopic echo /joint_states
```
And then, I move the ur10 arm with moveit, but still no valid velocities in the /joint_states topic can be found.
Can you help me?
Thank you very much.
GeWei

fmessmer, 2018-05-30:
The `ur10_moveit_config demo.launch` does not actually command the robot in the simulation!
It's basically meant to test the moveit config with fake controllers!
See the **Usage with Gazebo Simulation** part in the README about how to use MoveIt with a simulated robot!

geweihgg, 2018-05-30:
Hi, @fmessmer ,
I've modified the demo.launch file and the controllers.yaml, so that moveit can control the ur10 in gazebo, and ur10 moves as what the moveit plans.
Um... so, I think maybe other problem occurs, or my usage is not correct. I've tried to figure out why no  valid velocities there, but gained nothing.
Can you help me？
Thank you very  very much.
GeWei

fmessmer, 2018-05-30:
Why modify, when this repo already provides a correct way to use moveit_config!

geweihgg, 2018-05-30:
Hi, @fmessmer,
Um... I've just commented out the fake joint_state_publisher, and set fake_execution=False, then, set the name of controller in controllers.yaml to "manipulator".
GeWei

fmessmer, 2018-05-30:
Recommendation:
Undo the changes and use as documented!

However, I think I get the same behavior in Kinetic...
But I think then this is rather related to the `gazebo_ros_control` plugin and/or the physics of the UR urdf...as those are mainly responsible for the state reported from gazebo.

Can't help you any further here...I just wanted to point you to the correct usage of the moveit_config packages!

geweihgg, 2018-05-30:
Hi，
Thank you for your recommendation. I’ll try my best to fix this~
And maybe I’ll leave this issue open, waiting for someone who has fixed this.
Thank you very much!
GeWei

gavanderhoorn, 2018-05-30:
@geweihgg: as @fmessmer writes: don't edit the MoveIt configs or the other pkgs for this. It should not be needed.

The last section of the [readme](https://github.com/ros-industrial/universal_robot/blob/ad740e0dd028cd9614b869d306dfc2da402cbc46/README.md) (the *MoveIt! with a simulated robot* section) documents how to start things.

gavanderhoorn, 2018-05-30:
As to the issue you are encountering: I just tested this and I'm also not seeing the joint velocities I'm expecting, but I'm also not necessarily convinced we're seeing only 'noise'.

See this plot fi:

[image]

I just started Gazebo (`v2.4.15` on Indigo) and used `rqt_joint_trajectory_controller` to move individual joints around. If this was noise I would've expected far more randomness / chaotic behaviour. Instead I'm seeing something which *may* be a very downscaled version of what the data should actually be.

I'm not saying there isn't a problem here, just noting what I observed.

As to a solution: at this point I believe this is a problem in `gazebo_ros_control`, not in the `ur_description` package. As I noted earlier, we're just users here, and I don't believe we're doing any strange things.

I would suggest to open an issue over at [ros-simulation/gazebo_ros_pkgs/issues](https://github.com/ros-simulation/gazebo_ros_pkgs/issues) to report this. Be sure to refer back to the issue you posted here.

gavanderhoorn, 2018-05-30:
As for now I believe this is not something actionable for this repository, I'm going to close it.

@geweihgg: if there are any new developments, please re-open and update us.

closed 2018-05-30 by gavanderhoorn