ros-industrial/universal_robot #357: How to add calibration.conf data from controller to URDF for maximum accuracy

tragolf, 2018-06-26:
Hi,
I try to implement the calibration data from the UR-controller to the URDF-File. The goal is to reduce the discrepancy between the position of tool0 in MoveIt and the tool0_controller (2-3mm).
I copied the calibration.conf data from the controller. The content of the file is:
```
[mounting]  
delta_theta = [ 2.07750978653505926e-05, 0.96821571555565844, -1.85685385014877657, 0.888647901450376643, 5.66106814887767484e-05, -7.46310764096455082e-05]
delta_a = [ 0.000393621049984383038, 0.184093300397451837, 0.14493953300732923, -6.43297403642304399e-05, 7.76709040084077299e-06, 0]
delta_d = [ 4.00263097760228392e-05, 278.244144925623573, -183.695513914913732, -94.5471513669751289, 0.000304782681775556119, 0.000123737941387466743]
delta_alpha = [ -0.000271624944216508624, 0.00125856505456124258, -0.00322052070697146503, 0.00065153016530383745, -0.00058700465721117645, 0]
```

The following properties are defined in the URDF-File from ur_description of UR5:

```
<xacro:property name="ur5_d1" value="0.089159" />
<xacro:property name="ur5_a2" value="-0.42500" />
<xacro:property name="ur5_a3" value="-0.39225" />
<xacro:property name="ur5_d4" value="0.10915" />
<xacro:property name="ur5_d5" value="0.09465" />
<xacro:property name="ur5_d6" value="0.0823" />
```

How can I implement the calibration data in the URDF-File? Just adding the delta values to the values from the URDF-File results in crazy numbers.

gavanderhoorn, 2018-06-26:
Adding the deltas to the values in the urdf won't work, as you've experienced, as all those values are in metres and some of the offsets in the `calibration.conf` file are in the hundreds of metres (such as those in the `delta_d` line).

This topic has come up during a discussion we've had with UR, about exactly this topic.

Currently, the values from the `.conf` file cannot be directly used, and getting them in a form where we can use them is something that is being looked at.

> The goal is to reduce the discrepancy between the position of tool0 in MoveIt and the tool0_controller (2-3mm).

2 to 3mm difference between the two frames is actually quite good. I've seen differences up to 15mm (but that was with high loads and fast motions).

tragolf, 2018-06-26:
> Currently, the values from the .conf file cannot be directly used, and getting them in a form where we can use them is something that is being looked at.
So you can't give any hinds how to use them at the moment?

gavanderhoorn, 2018-06-26:
I've sent an email to my contacts to get some first hand information. If you need an answer right now, I wouldn't have anything for you. But if you can wait, we might.

tragolf, 2018-06-26:
I can wait, thank you very much!

urrsk, 2018-07-12:
Your main challenge is that you are trying to use the Denavit–Hartenberg (DH) parameters as 6D poses with 6 degrees of freedom (DOF), but DH only specifies 4 DOF.
DH parameters describe where a rotation axis is placed relative to its parent. So by using the parameters, you will get a point on the rotation axis, maybe 1km away depending on how parallel it is to the next joint.

The DH notation is a complete but singular notation. Singularities occur when a rotation axis is parallel to its parent axis. This is also the reason for the large numbers in the calibration file. 

So the task for you is to find a strategy to estimate 6 DOF using the 4 DOF DH has.

Suggestion:
You could project the base link origin towards the shoulders rotation axis. Using the new point on the shoulder axis and project that towards the elbow's rotation axis, and so forth... 

Note:
Multiple notations that are refered to as DH exist, however Universal Robots uses the original version which is identical to the one wiki describes: [https://en.wikipedia.org/wiki/Denavit%E2%80%93Hartenberg_parameters](https://en.wikipedia.org/wiki/Denavit%E2%80%93Hartenberg_parameters )

claudeHasler, 2018-09-06:
Hi, I'm currently facing the same issues regarding the units of the data in calibration.conf. Has any progress been made in this direction?

cpt-yoshi, 2018-09-12:
I have tried the suggestion @urrsk. Below is a sample of urdf for calculating the joints from dh parameters. Without the calibration applied, the <i>ee_link</i> pose is the same as the original ur3.urdf.xacro file. 

When applying the calibration, <i>tool0</i> link is within range for Y and Z (compared to tool0_controller). But for X there is a large offset of 11.7 meters. The DH convention used is the same as the ur3.urd.xacro file. The pose of the robot is HOME (vertically up). When moving the joints in URSim, the offset from tool0 to tool0_controller is kept, except for the elbow joint.  (To better visualize the joints in RViz, I divide d2, d3 and d4 from the calibration by 100) 

Before digging deeper, I have a couple of questions @urrsk :
- In the calibration file, are <b>delta_theta</b> and <b>delta_alpha</b> in radians?
- In the urd file, is the value of  <b>shoulder_offset</b> and <b>elbow_offset</b> necessary? My guess is that this information is already contained in the calibration file.
- Should the DH convention follow the universal robot  [found here](https://www.universal-robots.com/how-tos-and-faqs/faq/ur-faq/actual-center-of-mass-for-robot-17264/)? I presume that it should.
- Should the calibration offsets be added or substracted to the default DH paramters? I presume it should. 

<details><summary>ur3_calibration.urdf.xacro</summary>

```xml
<?xml version="1.0"?>
<robot xmlns:xacro="http://ros.org/wiki/xacro">

<!--
  Author: Felix Messmer
-->

  <xacro:include filename="$(find ur_description)/urdf/ur.transmission.xacro" />
  <xacro:include filename="$(find ur_description)/urdf/ur.gazebo.xacro" />

   <xacro:macro name="cylinder_inertial" params="radius length mass *origin">
    <inertial>
      <mass value="${mass}" />
      <xacro:insert_block name="origin" />
      <inertia ixx="${0.0833333 * mass * (3 * radius * radius + length * length)}" ixy="0.0" ixz="0.0"
        iyy="${0.0833333 * mass * (3 * radius * radius + length * length)}" iyz="0.0"
        izz="${0.5 * mass * radius * radius}" />
    </inertial>
  </xacro:macro>

  <xacro:macro name="ur3_robot" params="prefix joint_limited
     shoulder_pan_lower_limit:=${-pi}    shoulder_pan_upper_limit:=${pi}
     shoulder_lift_lower_limit:=${-pi}    shoulder_lift_upper_limit:=${pi}
     elbow_joint_lower_limit:=${-pi}    elbow_joint_upper_limit:=${pi}
     wrist_1_lower_limit:=${-pi}    wrist_1_upper_limit:=${pi}
     wrist_2_lower_limit:=${-pi}    wrist_2_upper_limit:=${pi}
     wrist_3_lower_limit:=${-pi}    wrist_3_upper_limit:=${pi}

     offset_a1:=0.0000205594635957679934
     offset_a2:=0.00150870902372793325
     offset_a3:=0.000793751111829499312
     offset_a4:=-0.0000193228507075169079
     offset_a5:=0.0000173718501728632923
     offset_a6:=0

     offset_d1:=0.0000135963461434518429
     offset_d2:=-110.351573314082572
     offset_d3:=123.970336119564919
     offset_d4:=-13.6188398018855246
     offset_d5:=0.0000190738707296911336
     offset_d6:=0.000477070250318664679

     offset_alpha1:=0.0000466705391628519095
     offset_alpha2:=0.000243411282863556191
     offset_alpha3:=0.00131412175387093889
     offset_alpha4:=0.000546787660626213423
     offset_alpha5:=-0.000984232803121987132
     offset_alpha6:=0

     offset_theta1:=0.0000278890969334113371
     offset_theta2:=-0.110482141198915154
     offset_theta3:=0.194449868933029041
     offset_theta4:=-0.0840284967969967267
     offset_theta5:=0.0000609090612742423896
     offset_theta6:=-0.00000396900258592857595">
     
     <!--
[... 474 lines cut ...]
    <!-- ROS base_link to UR 'Base' Coordinates transform -->
    <link name="${prefix}base"/>
    <joint name="${prefix}base_link-base_fixed_joint" type="fixed">
      <!-- NOTE: this rotation is only needed as long as base_link itself is
                 not corrected wrt the real robot (ie: rotated over 180
                 degrees)
      -->
      <origin xyz="0 0 0" rpy="0 0 ${-pi}"/>
      <parent link="${prefix}base_link"/>
      <child link="${prefix}base"/>
    </joint>

    <!-- Frame coincident with all-zeros TCP on UR controller -->
    <link name="${prefix}tool0"/>
    <joint name="${prefix}wrist_3_link-tool0_fixed_joint" type="fixed">
      <origin xyz="${x6} ${z6} ${y6}" rpy="${r6-pi} ${p6-pi/2} ${yaw6-pi/2}"/>
      <parent link="${prefix}wrist_3_link"/>
      <child link="${prefix}tool0"/>
    </joint>

  </xacro:macro>
</robot>
```

</details>

urrsk, 2018-10-12:
> I have tried the suggestion @urrsk. Below is a sample of urdf for calculating the joints from dh parameters. Without the calibration applied, the ee_link pose is the same as the original ur3.urdf.xacro file.
> 
> When applying the calibration, tool0 link is within range for Y and Z (compared to tool0_controller). But for X there is a large offset of 11.7 meters. The DH convention used is the same as the ur3.urd.xacro file. The pose of the robot is HOME (vertically up). When moving the joints in URSim, the offset from tool0 to tool0_controller is kept, except for the elbow joint. (To better visualize the joints in RViz, I divide d2, d3 and d4 from the calibration by 100)

> 
> Before digging deeper, I have a couple of questions @urrsk :
> 
> * In the calibration file, are delta_theta and delta_alpha in radians?

Yes, the theta and alpha angles is in radians.

> * In the urd file, is the value of  shoulder_offset and elbow_offset necessary? My guess is that this information is already contained in the calibration file.

No. Again, the DH only represent the rotation axis in the kinematic chain. Not were the physically joint house are placed on the rotation axis.

Challenge: Find a strategy to place that offset so it is aligns the geometry representing with the kinematics for e.g. the elbow joint. 

Suggestion: Find the point on the elbow's rotation axis where the base's rotation axis intersect.

> * Should the DH convention follow the universal robot  [found here](https://www.universal-robots.com/how-tos-and-faqs/faq/ur-faq/actual-center-of-mass-for-robot-17264/)? I presume that it should.

Yes, it also follow UR's own description. If it does not, let me know!

> * Should the calibration offsets be added or substracted to the default DH paramters? I presume it should.

Yes, the calibration offsets should be added to the default DH parameters.

> 
> ur3_calibration.urdf.xacro

cpt-yoshi, 2019-03-29:
I have successfully integrated the offset from my calibration file so that it aligns with the kinematic used in the urdf (validating using Matlab). This [tutorial](http://wiki.ros.org/Industrial/Tutorials/Create%20a%20URDF%20for%20an%20Industrial%20Robot) recommands this convention:

> The joint origin is specified by a full transform. However, in order to take advantage of some libraries, the joint orientation (roll, pitch, yaw) is limited to a single rotation.

Is this requirement only necessary for a revolute joint? Can I use a full transform for fixed joints?
This requirement will add approximately 9 dummy joints and dummy links.

gavanderhoorn, 2019-03-29:
> > The joint origin is specified by a full transform. However, in order to take advantage of some libraries, the joint orientation (roll, pitch, yaw) is limited to a single rotation.
> 
> Is this requirement only necessary for a revolute joint? Can I use a full transform for fixed joints?

it's not really a requirement, but more of a suggestion.

In any case: it only applies to revolute joints.

gavanderhoorn, 2019-03-29:
Please note that the work by Universal Robots announced at [ROS-Industrial EU Conference '18](https://rosindustrial.org/events/2018/12/11/ros-industrial-conference-2018) includes incorporating the calibration of individual robots into the URDFs/XACROs.

cpt-yoshi, 2019-04-02:
I have successfully integrated the offset in the URDF file by keeping the same axis joint rotation. There are some errors between my offsets values  **versus** the calibration file :

- Position:       A mean error of **1e-12** of the workspace (Minimum of **1e-09**) 
- Orientation:  A mean error of **1e-13** of the workspace (Minimum of **1e-10**)

I think the errors comes from numerical round-off (I could compare with symbolic arithmetic to see if it's the case). But with precision error as low as nanometers, I don't think it matters...

If someone could post a calibration file from a ur5 and a ur10, then I could check if my algorithm works. I validated my eef with the _tool0_controller_ floating joint in RVIZ. I will post my script in a few days (Must do some cleanup first!). I guess it's not worth trying to automate this process if Universal Robots is going to implement it.

gavanderhoorn, 2019-04-02:
Nice to hear about your results. It would seem to be in about the range that @fmauch reports over in #414.

> I guess it's not worth trying to automate this process if Universal Robots is going to implement it.

Perhaps you could compare notes with @fmauch and post a comment in #414 to compare your approaches?

cpt-yoshi, 2019-04-10:
Here is the [repo](https://github.com/cpt-yoshi/universal_robot/tree/calibration) that contains my algorithm. The **readme** explains how to add your calibration file in the URDF. I also explain how to validate your calibrated URDF  in RViz so that you can compare your eef_link with tool0_controller.

twhitney1126, 2019-04-22:
@cpt-yoshi I'm a bit confused. Can you explain where the values for lines 121 through 126 came from?
Especially, T4_5NC_URDF and T5_6NC_URDF. Is there a chance that you mistyped T4_5NC_URDF as 0.08505 instead of 0.08535?

cpt-yoshi, 2019-04-23:
@twhitney1126 These coordinate frames values came from the original URDF file: **URX.urdf.xacro**. These TF are needed to compute the new calibrated joints to be as close as possible with the joints in the original URDF so that it doesn't affect to much the visual and collision meshes. For the UR3, the frame coordinate T4_5NC_URDF is the transform from _wrist_1_joint_ to _wrist_2_joint_ that has a value of **wrist_1_length**:

```xml
    <xacro:property name="d4" value="0.11235" />
    <xacro:property name="d5" value="0.08535" />
    <xacro:property name="d6" value="0.0819" />

    <!-- Arbitrary offsets for shoulder/elbow joints -->
    <xacro:property name="shoulder_offset" value="0.1198" />  <!-- measured from model -->
    <xacro:property name="elbow_offset" value="-0.0925" /> <!-- measured from model -->

    <!-- link lengths used in model -->
    <xacro:property name="shoulder_height" value="${d1}" />
    <xacro:property name="upper_arm_length" value="${-a2}" />
    <xacro:property name="forearm_length" value="${-a3}" />
    <xacro:property name="wrist_1_length" value="${d4 - elbow_offset - shoulder_offset}" />
```
So the wrist_1_length value gives: 0.11235 - (-0.0925) - 0.1198 = 0.08505. The coordinate frame T5_6NC_URDF gives 0.08535.

I will update (tonight) the code to reflect the names of the joints given in the URDF so that there is less ambiguity.

cpt-yoshi, 2019-04-23:
The repo has been updated along with the _readme_. The joint names in the table now reflects the URDF convention naming convention. In RViz you now must compare _tool0_ with _tool0_controller_. The _ee_fixed_joint_ respects the non-calibrated joint axe.

I have added an estimation error table that compare the newly calibrated joints vs the original URDF ones. The max position error is 364 micro-meters and a max orientation angle error of 1.16 mRad. With these max errors, I don't think this will affect **inertias** for simulation but I haven't tested this in Gazebo.

[image]

aperaid, 2019-04-25:
> Here is the [repo](https://github.com/cpt-yoshi/universal_robot/tree/calibration) that contains my algorithm. The **readme** explains how to add your calibration file in the URDF. I also explain how [...]

Pardon me, I tried the matlab formula (UR5 robot), then compare:
rosrun tf tf_echo /base /tool0_controller vs rosrun tf tf_echo /base /tool0
But the result is even more inaccurate.
(left is tool0_controller, right is tool0)
[image]
This time, all X, Y, Z are off.

What I did:
1. Line 13 to 39: copy the UR5 default d-h parameters
```
a_nc = [0  
       -0.425
       -0.39225
        0  
        0
        0]';

d_nc = [0.089159 
        0 
        0 
        0.10915 
        0.09465
        0.0823]';

alpha_nc = [pi/2  
            0  
            0  
            pi/2 
           -pi/2
            0]';

theta_nc = [0
            0
            0
            0
            0 
            0]';
```
2. Line 47 to 73: copy the calibration.conf value
- delta_theta to theta_c
- delta_a to a_c
- delta_d to d_c
- delta_alpha to alpha_c
3. Line 121 to 126: change the RPY value following the joints
```
TB_1NC_URDF   = [rpy2rot(0, 0, 0)       [0, 0, 0.089159]';        0 0 0 1];
T1_2NC_URDF   = [rpy2rot(0, pi/2, 0)    [0, 0.13585, 0]';        0 0 0 1];
T2_3NC_URDF   = [rpy2rot(0, 0, 0)       [0, -0.1197, 0.42500]'; 0 0 0 1];
T3_4NC_URDF   = [rpy2rot(0, pi/2, 0)    [0, 0, 0.39225]';       0 0 0 1];
T4_5NC_URDF   = [rpy2rot(0, 0, 0)       [0, 0.093, 0]';       0 0 0 1];
T5_6NC_URDF   = [rpy2rot(0, 0, 0)       [0, 0, 0.09465]';       0 0 0 1];
```
4. Run the matlab, get the value, paste it to each joints in the URDF
5. Copy this URDF into ur_description "ur5.urdf.xacro"
Testing:
```
roslaunch ur_modern_driver ur5_bringup_joint_limited.launch robot_ip:=x.x.x.x
roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch
rosrun tf tf_echo /base /tool0_controller
rosrun tf tf_echo /base /tool0
```

Any suggestions?

cpt-yoshi, 2019-04-26:
I haven't tested it on a UR5 and UR10, so there might be an error in the algorithm. The parameters that you modified are all good.

Can you validate that the **Table 2** given by Matlab has approximatively a mean error of 1e-12 for position and 1e-09. If these values are good that means the error is somewhere in the URDF.

If you give me your calibration file and your Xacro, I can help you out.

fmauch, 2019-04-29:
Sorry for hooking into this discussion so late, I was away the last weeks... As @gavanderhoorn  mentioned above, I'm currently developing a calibration routine for UR robots as part of the new UR RTDE driver.

For this I created a c++-only calibration routine that "corrects" the elbow and wrist locations to be as close to the uncalibrated model as possible. In #414 I raised a couple of questions on how to integrate this into the description lying in this repository.

As far as I can see, you (@cpt-yoshi) decided to calculate all 6 DOF for each joint and manually enter this into the calibration file. As I would like to automate this a little more, it would be good to make the xacro file parametrizable, don't you think?

I currently go for a separate file containing the calibration information, which is currently a DH-like notation with 4 parameters per joint plus two extra parameters to specify the additional rotation introduced by the correction. When I have a look at both our implementations, it would be best to specify all xyz,rpy sets to the description. If I implemented that, I think it could be used with your algorithm, as well. What do you think?

aperaid, 2019-05-06:
> I haven't tested it on a UR5 and UR10, so there might be an error in the algorithm. The parameters that you modified are all good.
>
> Can you validate that the **Table 2** given by Matlab has [...]

Hi @cpt-yoshi, thank you for following up.

Attached:
[ur5_calibration_files.zip](https://github.com/ros-industrial/universal_robot/files/3147281/ur5_calibration_files.zip)

Table 2:
WorkspaceError; err_mean; err_std; err_max

    'Position'        4.49200401439215e-11    2.65005132924673e-11    8.40448298356843e-11
    'Orientation'     3.92104587027649e-10    7.73899643295251e-09    5.90701006363192e-07

Table 3:
Joint Name; pos_error; orientation_err

    'shoulder_pan_joint'                3.03043719878818e-20    0.00000000000000e+00
    'shoulder_lift_joint'               3.65123831653643e-04    3.07465540749986e-04
    'elbow_joint'                       3.73632344497944e-04    6.94828484597094e-04
    'wrist_1_joint'                     6.07777974220929e-04    2.57762587358672e-03
    'wrist_2_joint'                     1.56987328803007e-03    3.11324273067055e-03
    'wrist_3_joint'                     8.18786365965450e-04    2.74410403145109e-03
    'ee_fixed_joint'                    1.98625944262863e-03    2.74416125856863e-03
    'wrist_3_link-tool0_fixed_joint'    1.98625944262863e-03    2.74416125856863e-03

twhitney1126, 2019-05-06:
@cpt-yoshi I was able to write a generic C++ of this, that takes the two .conf files as inputs, parses the necessary values from each and creates a new URDF file. After implementation, I compared the UR known TCP position from the teach pendant and compared the result with what TF believed was the TCP and they were within 0.7 mm. Prior to the updated URDF, the error between the to was about 4.5 mm.

NOTE: The only two inputs I couldn't make generic were the measured offsets in the URDF and the specified joint rotations.

cpt-yoshi, 2019-05-07:
@aperaid I will also need your **safety.conf** file located in ~/.urcontrol. I can't initialize the simulated robot without it (In UR5 mode).

There seems to have no errors in the files you uploaded.

cpt-yoshi, 2019-05-07:
> I was able to write a generic C++ of this, that takes the two .conf files as inputs, parses the necessary values from each and creates a new URDF file. 

Good to hear!

> After implementation, I compared the UR known TCP position from the teach pendant and compared the result with what TF believed was the TCP and they were within 0.7 mm. Prior to the updated URDF, the error between the to was about 4.5 mm.

The 0.7 mm mismatch seems a lot compared to what the algorithm reports. The max error calculated from a UR5 is in the range of 8.4e-11 meters... The algorithm dosen't validate the entire workspace but I would assume the error to not go below nano meters.

cpt-yoshi, 2019-05-07:
@fmauch 
> As far as I can see, you (@cpt-yoshi) decided to calculate all 6 DOF for each joint and manually enter this into the calibration file. As I would like to automate this a little more, it would be good to make the xacro file parametrizable, don't you think?

When @gavanderhoorn told me that Universal Robot was planning to work on this, I decided to released what I had at that point. I would have eventually automated this step with c++/python with a parametrizable xacro file or something that generates a urdf file.

> When I have a look at both our implementations, it would be best to specify all xyz,rpy sets to the description. If I implemented that, I think it could be used with your algorithm, as well. What do you think?

Personally, I think it's best to specify all 6 degree of freedom. But as pointed out by @gavanderhoorn, libraries like IKFast tend to like only one rotation at at time. To fix this, dummy joints would have to be added in the URDF because there are offsets in almost all rotations (r, p, y) of each joints. 

On a side note, I would also change the naming links/joints in the URDF to reflect with this [REP](http://gavanderhoorn.github.io/rep/rep-xxxx.html).

gavanderhoorn, 2019-05-07:
> But as pointed out by @gavanderhoorn, libraries like IKFast tend to like only one rotation at at time. To fix this, dummy joints would have to be added in the URDF because there are offsets in almost all rotations (r, p, y) of each joints.

just to clarify: this comment was based on my experience with IKFast. It could be that things have changed or improved. It'd be worth making sure the advice is still relevant.

If you can provide me with a calibrated urdf I'll see if I can run the IKFast generator on it.

twhitney1126, 2019-05-07:
@cpt-yoshi @gavanderhoorn I have tested both the forward kinematics and inverse kinematics with the updated URDF. There still seems to be a few issues that show up... After applying the updated URDF, I published the expected pose and the actual robot pose and they never seemed to match exactly (same mm error), so I dug a little deeper. I found that ur_kinematics/src/ur_kin.cpp has hard-coded values for the UR3, 5, 10 (and assumes exactly orthogonal joints) that are created at compile time. This is why the robot never gets to the expected location.

gavanderhoorn, 2019-05-07:
Solvers that do not take runtime parameters into account will obviously not benefit from the calibration improvements.

FK/IK solvers that are capable of doing that are `robot_state_publisher`, `trac_ik` and `KDL`.

cpt-yoshi, 2019-05-07:
> If you can provide me with a calibrated urdf I'll see if I can run the IKFast generator on it.

@gavanderhoorn The file ur3_calibration_robot.urdf.xacro located in this [repo](https://github.com/cpt-yoshi/universal_robot/tree/calibration) has my personnal calibration offsets. The repo also contains my personnal config files for URSim located [here](https://github.com/cpt-yoshi/universal_robot/tree/calibration/ur_description/calibration).

> I found that ur_kinematics/src/ur_kin.cpp has hard-coded values for the UR3, 5, 10 (and assumes exactly orthogonal joints) that are created at compile time. This is why the robot never gets to the expected location.

I only tested with KDL... That would also explain the inaccuracy of @aperaid. Can you confirm that you are using ur_kinematics as your kinematic solver?

> Solvers that do not take runtime parameters into account will obviously not benefit from the calibration improvements.

I have update the Readme to add the current limitations with ur_kinematics.

hariharan82, 2019-06-18:
@cpt-yoshi , I am working with the UR10 and I used you algorithm to generate the modified/calibrated DH. after updating the urdf with those parameters, i am not getting the FK to match with what the controller says. Could you please run your algorithm with my calibration offsets and guide me as to what to change? thanks a lot in advance.
```delta_theta = [ -7.68578854641674555e-05, -0.0484988195154549828, 0.0773589887018100442, -0.0286841179499151347, -6.23968602647445076e-05, -1.36814605310694358e-06]
delta_a = [ -7.06689796862263305e-05, 0.000473292415404835687, 0.00142607850210341169, -9.47264317413558358e-06, 8.52074827089885432e-05, 0]
delta_d = [ 0.000945515566846549804, 5.43382695506826963, -8.0663748652800713, 2.63227402564373003, 9.0280898773970053e-05, 0.000204171581686854453]
delta_alpha = [ 0.000255797545358982248, -0.00546750558962930643, -0.00623286337061196728, 0.000170403695184617732, 0.000277199666766536623, 0]
```

raschoff, 2019-07-19:
Can someone point me out to where would be the algorithm to generate the yaml supporting file?
Even better, is there a step list anywhere so we can get from the the calibration.conf to the URDF?
I am not even using ROS right now, but I could install it just to get the calibrated URDF.

This is my calibration file the UR5 CB3 I am working with.
`[mounting]
delta_theta = [ 6.78585256882133101e-05, -1.31462654806621093, 1.25389332785503549, 0.0607419221392160105, 2.73542834480446065e-05, 6.90070973687733853e-06]
delta_a = [ -1.97986926564499874e-05, 0.317287112349698974, 0.000922802555250346046, -4.89180725661515959e-05, 0.000163041731739411739, 0]
delta_d = [ -0.000123193809323934866, 834.453714740628129, -842.724160504791939, 8.27180353984201844, 0.000119684949253073825, 0.000364100515684556703]
delta_alpha = [ -0.000962270916070817606, -0.000492947781173158476, 0.00288238586893947409, -0.000911038884981119068, 0.00037680732967237951, 0]
joint_checksum = [ 0xd626f97a, 0xfb9a0710, 0x14f38a86, 0x294070dc, 0x1a77ab80, 0xf4041bd9]
calibration_status = 2 # 0 == notInitialized / 1 == notLinearised / 2 == Linearised
joint_raw_offset = [ 0xd4530d, 0xd3aee0, 0x54f2c6, 0xfdc5c2, 0xfc756d, 0x7c6f2f]
joint_selftest_data_crc = [ 0xfbb628a9, 0x390f1ac9, 0x6c87c69f, 0xcee51c, 0x64750aa7, 0xc4fa1710]`

fmauch, 2019-07-19:
The script is part of a new driver suite that is currently in a closed beta state. It will be released soon(ish), hopefully.

nicolasCheron, 2019-09-23:
Hello, In order to get an accurate model of my UR10 in simulation, I moved the robot in hundreds of positions and get the position in Cartesian and joint space. Then I used scipy to find the DH parameters that minimized the error between the cartesian position and the position from the direct geometrical model. I limited the search around the theoretical DH parameters. I have now this parameters :
 `DeltaX :
d = [7.07489864e-04,-1.79827174e-04,-1.76065257e-04,-2.05058836e-04,-2.49932350e-05,4.13236499e-07]
a = [6.09568665e-05,-2.43622129e-04,1.15715437e-03,3.50628841e-05,2.02848136e-05,1.44781964e-06]
alpha = [1.74431908e-04,-2.88575826e-04,-9.55831763e-04,4.34453688e-04,-1.55413778e-04,-4.40614317e-05]
theta = [5.18360443e-03,-1.18625151e-03,2.53951697e-03,-3.99284086e-03,3.91415132e-02,1.98777510e-03]` 

The error is less than 1 mm and 0.001 rads on the three rotation angles. 
These values can be used to get the inverse kinematic in closed form (it did not work with the delta_d in hundreds of meters). 

But I don't get how to create the urdf from the DH parameters as the coordinates  of the robot are not the same (all rotations are not on z axis). Does someone know how to fix this ?

fmauch, 2019-09-25:
The soon(ish) mentioned above comes closer probably being somewhere in October.

gavanderhoorn, 2019-11-23:
With `ur_robot_driver` now released, the calibration conversion infrastructure is now also available.

Therefore I'm closing this issue.

Refer to the [announcement on ROS Discourse](https://discourse.ros.org/t/deprecation-of-ur-modern-driver/11629).

For further information on how `ur_robot_driver` uses calibration data and how to extract it from a controller, refer to the documentation available in the new repository.

closed 2019-11-23 by gavanderhoorn