ros-industrial/universal_robot #370: Unify UR3/UR5/UR10 MoveIt! configurations

zultron, 2018-07-25:
The three UR robot MoveIt! configurations are almost identical except for small differences in the `ur*_moveit_config/` directories:
- The UR10 `max_velocity` setting in `config/joint_limits.yaml` is different
- There are a few (I suspect insignificant) `<disable_collisions/>` differences in `config/ur*.srdf`
- Different kinematics plugins in `config/kinematics.yaml` (interestingly, the plugin sources are unified)
- Pointers to different `urdf/ur*_robot.urdf.xacro` files in the `ur_description` package
  - Those files, too, are quite close except for mass, cog and kinematics variable values
- Trivial differences like `$(find ur*_moveit_config)` and `<arg [...] value="ur*"/>`

I believe all of these differences could be parameterized so that the files could be unified.  Why aren't they?

Possible reasons I can think of to unify:
- DRY:  Updates only need to be applied once, and needn't be duplicated in three places; this reduces work and ensures that fixes to one robot model are applied to all models
- Simplify adding new models:  In #210, @fmessmer unified some `.xacro` files in the `ur_description` package, citing a maintainability advantage

Possible reasons I can think of *not* to unify:
- Breaking the setup assistant:  The setup assistant makes it easier to keep up with updates to MoveIt!
- Breaking backward compatibility:  In #232, @gavanderhoorn pointed out that the changes in #210 could break compatibility with existing configurations
- No compelling need:  Things work the way they are now, there's no unbearable maintenance burden maintaining separate files, so why bother with this exercise?

What's missing from this big picture view?

gavanderhoorn, 2018-07-25:
> Breaking the setup assistant: The setup assistant makes it easier to keep up with updates to MoveIt!

this -- together with: "it currently works" -- is the main reason why we don't merge moveit configurations.

The MSA makes creating a config 3 mins of clicking next-next-next, and in addition, MoveIt configs have very low reusability, unless you happen to have the *exact* same robot and environment setup. Any changes in environment, toolframes, manipulator groups (add an end-effector, etc) will require a rerun of the MSA.

So the configs we have in this repository, and all the other MoveIt configs we provide in other repositories are essentially mostly examples / convenience packages: you can use them to play a bit with the model and to do some first tests, but any serious use of MoveIt and any robot will require you to create a application-specific configuration anyway.

Merging MoveIt configurations would be nice, but has not been worth it so far (and the three variants we have here in `universal_robot` is not even that many: see [fanuc_lrmate200ic_support](http://wiki.ros.org/fanuc_lrmate200ic_support), that package supports 8 variants of the same robot).

zultron, 2018-07-25:
Thanks for the thoughtful reply!  It answers my questions perfectly.

FWIW, here's a script I used to translate all e.g. `ur3` to `urX` to make the comparison easier:
```
for bot in 3 5 10; do
    # Make file names generic
    for f in \
        config/ur${bot}.srdf \
        launch/ur${bot}_moveit_planning_execution.launch \
        launch/ur${bot}_moveit_controller_manager.launch.xml \
        launch/ur${bot}_moveit_sensor_manager.launch.xml; do
            mv ur${bot}_moveit_config/${f} ur${bot}_moveit_config/${f/${bot}/X}
    done
    # Make file contents generic
    find ur${bot}_moveit_config -type f -exec sed -i \
        -e "s/UR${bot}/URX/g" -e "s/ur${bot}/urX/g" {} +
done
```

Closing this issue.

closed 2018-07-25 by zultron

gavanderhoorn, 2018-07-25:
np.

It's a good question btw, and you're not the first to ask it. See https://github.com/ros-industrial/fanuc/issues/125 fi.