ros-industrial/universal_robot #373: launch errors: No matching hardware interface found

stevensu1838, 2018-07-31:
When I run roslaunch ur_gazebo ur5.launch in the terminal, I got the following errors and the robot doesn't  look right in Gazebo. Can you please give a favor? Thanks a lot. 

[```
FATAL] [1533015918.919675549, 0.171000000]: No matching hardware interface found for 'hardware_interface/PositionJointInterface
[FATAL] [1533015918.919760922, 0.171000000]: Could not initialize robot simulation interface
```

![image](https://user-images.githubusercontent.com/25648734/43440053-ef20bacc-94e9-11e8-8943-87f246e512d1.png)

stevensu1838, 2018-07-31:
[image]

gavanderhoorn, 2018-07-31:
Which version of Gazebo are you both using?

stevensu1838, 2018-08-13:
I tried 

$ sudo apt-get install \
  ros-indigo-ur-gazebo \
  ros-indigo-ur5-moveit-config \
  ros-indigo-ur-kinematics

and 

$ sudo apt-get install ros-indigo-gazebo-ros-pkgs ros-indigo-gazebo-msgs ros-indigo-gazebo-plugins ros-indigo-gazebo-ros-control

Then source /opt/ros/indigo/setup.bash
It works for me

gavanderhoorn, 2018-08-13:
So things work for you now?

gavanderhoorn, 2018-08-26:
Ping.

I would still like to know how you installed the packages before, seeing as the dependencies weren't installed.

gavanderhoorn, 2019-06-12:
Closing due to inactivity.

closed 2019-06-12 by gavanderhoorn