ros-industrial/universal_robot #374: [melodic] UR5 Moveit model not showing correctly

Pro, 2018-08-03:
I am on a newly installed Ubuntu 18.04 with ROS Melodic.

I installed all the ROS packages with apt and then cloned the universal_robot repo into my workspace.

Then I run:
```sh
roslaunch ur_gazebo ur5.launch   
roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true
roslaunch ur5_moveit_config moveit_rviz.launch config:=true
```
which opens the following windows:

[image]

[image]

As you can see in RViz the robot looks somehow broken. If I try to plan a path, it always fails since the robot is already colliding.
Do you have any clues how to fix this?

Thanks!

gavanderhoorn, 2018-08-03:
This could be a Gazebo 9 incompatibility issue.

Could you please provide us with the version of Gazebo you have running and a single message from the `/joint_states` topic.

Pro, 2018-08-06:
Here are the versions I have installed:

```
ros-melodic-gazebo-ros-control/bionic,now 2.8.4-0bionic.20180710.223050 amd64 
gazebo9/bionic,now 9.0.0+dfsg5-3ubuntu1 amd64 [installed,automatic]
```

And the output of joint_states is:

```
---
header: 
  seq: 2638
  stamp: 
    secs: 53
    nsecs: 413000000
  frame_id: ''
name: [elbow_joint, shoulder_lift_joint, shoulder_pan_joint, wrist_1_joint, wrist_2_joint,
  wrist_3_joint]
position: [8.404066400302668e-05, 0.013313130523400218, 0.00048356653853609544, 0.0004878671203663032, -5.116296311058477e-06, 3.773704196063932e-05]
velocity: [0.00019241551370976356, 0.04266308535101533, -0.0001239670664260386, 0.0014172778802119534, -7.408810395117624e-06, 0.00011813872303687023]
effort: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```

IMHO the joint states message should be fine?

Here are the outputs of the launch commands:

**roslaunch ur_gazebo ur5.launch**

```
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://pc1-n-094:32775/

SUMMARY
========

PARAMETERS
 * /arm_controller/action_monitor_rate: 10
 * /arm_controller/constraints/elbow_joint/goal: 0.1
 * /arm_controller/constraints/elbow_joint/trajectory: 0.1
 * /arm_controller/constraints/goal_time: 0.6
 * /arm_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /arm_controller/constraints/shoulder_lift_joint/trajectory: 0.1
 * /arm_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /arm_controller/constraints/shoulder_pan_joint/trajectory: 0.1
 * /arm_controller/constraints/stopped_velocity_tolerance: 0.05
 * /arm_controller/constraints/wrist_1_joint/goal: 0.1
 * /arm_controller/constraints/wrist_1_joint/trajectory: 0.1
 * /arm_controller/constraints/wrist_2_joint/goal: 0.1
 * /arm_controller/constraints/wrist_2_joint/trajectory: 0.1
 * /arm_controller/constraints/wrist_3_joint/goal: 0.1
 * /arm_controller/constraints/wrist_3_joint/trajectory: 0.1
 * /arm_controller/joints: ['shoulder_pan_jo...
 * /arm_controller/state_publish_rate: 25
 * /arm_controller/stop_trajectory_duration: 0.5
 * /arm_controller/type: position_controll...
 * /gazebo/enable_ros_network: True
 * /joint_state_controller/publish_rate: 50
 * /joint_state_controller/type: joint_state_contr...
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 50.0
 * /robot_state_publisher/tf_prefix: 
 * /rosdistro: melodic
 * /rosversion: 1.14.2
 * /use_sim_time: True
[... 262 lines cut ...]
[ INFO] [1533537791.639166091]: rviz version 1.13.1
[ INFO] [1533537791.639211981]: compiled against Qt version 5.9.5
[ INFO] [1533537791.639224265]: compiled against OGRE version 1.9.0 (Ghadamon)
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[ INFO] [1533537792.172455630, 132.468000000]: Stereo is NOT SUPPORTED
[ INFO] [1533537792.173165167, 132.469000000]: OpenGl version: 4.6 (GLSL 4.6).
libpng warning: iCCP: known incorrect sRGB profile
[ INFO] [1533537795.448819144, 135.740000000]: Loading robot model 'ur5'...
[ WARN] [1533537795.448876936, 135.740000000]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1533537795.448913503, 135.740000000]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1533537795.476073929, 135.768000000]: Loading robot model 'ur5'...
[ WARN] [1533537795.476101482, 135.768000000]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1533537795.476112052, 135.768000000]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1533537795.543761234, 135.835000000]: Starting scene monitor
[ INFO] [1533537795.545602520, 135.837000000]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1533537795.788131702, 136.079000000]: Constructing new MoveGroup connection for group 'manipulator' in namespace ''
[ INFO] [1533537796.937293591, 137.227000000]: Ready to take commands for planning group manipulator.
[ INFO] [1533537796.937350589, 137.227000000]: Looking around: no
[ INFO] [1533537796.937376834, 137.227000000]: Replanning: no
[ WARN] [1533537796.951931285, 137.241000000]: Interactive marker 'EE:goal_ee_link' contains unnormalized quaternions. This warning will only be output once but may be true for others; enable DEBUG messages for ros.rviz.quaternions to see more details.

```

Please let me know if you need any more data/info!

Pro, 2018-08-06:
If I enable the `tf` visualization in rviz, it also shows the correct frame position, only the robot model is not correct:

[image]

Pro, 2018-08-06:
Note that I posted a more general question here https://answers.ros.org/question/299709/where-does-rvizmoveit-get-the-link-position/
I will keep both discussions in sync!

gavanderhoorn, 2018-08-06:
@Pro wrote:
> Note that I posted a more general question here https://answers.ros.org/question/299709/where-does-rvizmoveit-get-the-link-position/
I will keep both discussions in sync!

That was not really necessary: there is no magic topic, it's all TF.

Pro, 2018-08-06:
> That was not really necessary: there is no magic topic, it's all TF.

Ok, sorry! I wanted to keep the general discussion separated from the issue here, since I'm new in RViz+Gazebo, the question on answers.ros.org was more a general understanding question.

Could it be a universal_robot package configuration issue that RViz is not taking the correct tf data?

gavanderhoorn, 2018-08-06:
At this point I don't know -- I've not migrated to Melodic.

have you changed anything in the `ur_description` package? Do things look normal if you run:

```
roslaunch ur_description ur5_upload.launch
roslaunch ur_description test.launch
```

The fact that the TF frames look normal seems to indicate this is purely a rendering issue.

gavanderhoorn, 2018-08-06:
The misconfigured fixed frame is expected, that is not a problem.

> It looks like the moveit configuration is also using some invalid fixed frame and thus the model is not shown correctly.

No, those are not related, and it's also not the same concept.

Please don't make statements about causes unless you've verified them.

gavanderhoorn, 2018-08-06:
I won't have time to look at this more today or tomorrow unfortunately.

If you could try and see whether you can determine what is going on, that would be great.

Otherwise I can probably take a look later.

Pro, 2018-08-06:
Not sure what I can change in the config for `fixed_base`... Unfortunately I'm a newbie in Gazebo+RViz+Ros Control so I really appreciate any help you could provide.

You can probably easily reproduce the issue by installing Ubuntu 18.04 in a VM and just install ros-melodic-desktop-full. Then cone this repo into a workspace.

ipa-nhg, 2018-08-06:
@Pro thanks for testing UR with melodic. 

Could you please take a screenshot of the collisions (the collisions are stl files instead of collada)? If the frames are right and the origins of the visual links are zero, looks like the problem is something with the 3D models(?).

I recommend just start rviz and the joint and robot state publisher like this:

```
roscd ur_description/urdf
roslaunch urdf_tutorial display.launch model:=ur5_robot.urdf.xacro gui:=true
```
Go to RobotModel disable "Visual enabled" and enable "Collision enabled" and please take a screenschot ( check also that the TF tag "Show axes" is enabled). Then press the button "Randomize" of the joint_state_publisher dialog and check RVIZ again.

did you test also the UR10 or the UR3?

Thanks a lot!!

Pro, 2018-08-06:
@ipa-nhg thanks for your help! Here some screenshots:

If I run exactly the commands you provide, the output is this:
[image]

So this looks OK to me.

Randomize also works as expected:

[image].

So it looks to me that the models are correct and the issue is together with gazebo.

UR10 has the same issue when executing it together with Gazebo:

```
roslaunch ur_gazebo ur10.launch limited:=true
roslaunch ur10_moveit_config ur10_moveit_planning_execution.launch sim:=true limited:=true
roslaunch ur10_moveit_config moveit_rviz.launch config:=true
```

[image]

ipa-nhg, 2018-08-06:
@pro Thanks a lot for the quick answer, can you please test this branch of my fork https://github.com/ipa-nhg/universal_robot/tree/MelodicHotFix ??

Please do not close the issue if this version solves the problem, it is just a Hot Fix and not a proper solution.

Pro, 2018-08-06:
Unfortunately it's still the same with your version. The only difference is that the robot does not include any color now:

[image]

Pro, 2018-08-06:
Could it be somehow related to an invalid fixed frame setting?

I.e. if I select an invalid fixed frame in RViz (without starting gazebo), the robot looks somehow similar to the broken one, when starting with gazebo. (as mentioned here: https://github.com/ros-industrial/universal_robot/issues/374#issuecomment-410637469)

So just a wild guess:
When executing `roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true limited:=true` it shows the warning:

```
[ INFO] [1533565707.758143775]: Loading robot model 'ur5'...
[ WARN] [1533565707.758858906]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1533565707.758873548]: No root/virtual joint specified in SRDF. Assuming fixed joint
```
And since the Scene Robot description is using the SRDF file for the `/robot_description_semantic` topic, there could be an issue with the SRDF file, that it does not define a valid root frame, and thus all the relative frame positions are set to 0,0,0.

Pro, 2018-08-07:
Yayyy, found the issue:

It is in more detail described here:
https://github.com/ros-planning/moveit/issues/690 and https://github.com/ros/urdfdom_headers/issues/41

If you have a german locale setting, i.e. `export LC_NUMERIC="de_DE.UTF-8"` and then start RViz, you see the broken image. If you manually set the locale to `export LC_NUMERIC="en_US.UTF-8"` and then start RViz, everything is fine.

Note that the PR https://github.com/ros/urdfdom_headers/issues/41 is not (yet) included in the current melodic release.

closed 2018-08-07 by Pro

gavanderhoorn, 2018-08-07:
Ah, the locale issue.

Good find.

Thanks for reporting back.

sapan-ostic, 2019-03-13:
How was the model appearing dark in gazebo solved? The RViz model loads up fine for me but Gazebo doesnt.

miguelprada, 2019-03-14:
Hi @sapan-ostic, you already posted a [similar question](https://github.com/ros-industrial/universal_robot/pull/377#issuecomment-470369050) in another thread. Please don't post duplicate questions in unrelated issues and follow up there if you'd like further clarifications.

Musyue, 2019-04-13:
@Pro Hi,I found the same error like you ,when i open bring up ur5 model in gazebo,it show a error position about the model,could you give me some information to fix this problem?