ros-industrial/universal_robot #375: ur_kinematics/UR5KinematicsPlugin failed

yt100323, 2018-08-08:
ros: indigo
os: ubuntu 14.04
my problem is that when i add a robotiq gripper to ur5, and then the ur_kinematics seems failed when using moveit to plan to a random pose while the bare ur5 model works fine

yt100323, 2018-08-09:
Thank you for you replay, I have generated a new srdf file like this[https://github.com/ECNURoboLab/ecnu_ur5_setup/blob/master/ecnu_ur5_setup_moveit_config/config/ecnu_ur5.srdf](url), and group "manipulator" do not add additional joint, what is the problem?

yt100323, 2018-08-09:
@gavanderhoorn

gavanderhoorn, 2018-08-13:
>  what is the problem?

I'm not sure: what is the problem now?

Note: the `chain` is defined as `base_link->ee_link` in [the original UR5 SRDF](https://github.com/ros-industrial/universal_robot/blob/2af9088dfa2671c8184900c0710e85ea54f2413f/ur5_moveit_config/config/ur5.srdf#L13).