ros-industrial/universal_robot #376: Cant integrate the code with Unity!

animant2, 2018-08-09:
Hey guys,
I've been trying to integrate the code for Inverse Kinematics for UR5 Robot just to fail each time I try it. 
I have taken the int inverse() function and converted it to C#. Parameters are ten times bigger in Unity space. So I have multiplied them all by 10, then when I give the code a Target Location and then extract its localToWorldMatrix, I get the angle options but they are mega wrong. It's been a while since I've started trying to solve the inverse kinematics for the UR5. My aim is just to give the controller a Target Point, then the Robot will move its End Effector to it. 
`public float VirtualInverse(GameObject Target, float[] q_sols)
    {
        int num_sols = 0;
        Matrix4x4 EndPose = new Matrix4x4();
        EndPose = Target.transform.localToWorldMatrix;
        float T00 = EndPose.GetColumn(0).x; float T01 = EndPose.GetColumn(1).x; float T02 = EndPose.GetColumn(2).x; float T03 = EndPose.GetColumn(3).x;
        float T10 = EndPose.GetColumn(0).y; float T11 = EndPose.GetColumn(1).y; float T12 = EndPose.GetColumn(2).y; float T13 = EndPose.GetColumn(3).y;
        float T20 = EndPose.GetColumn(0).z; float T21 = EndPose.GetColumn(1).z; float T22 = EndPose.GetColumn(2).z; float T23 = EndPose.GetColumn(3).z;`

The calculation part is the same with the c++ file. 
In the end of the Inverse function I have this code: 
`for(int k= 0; k < 2; k++)
                {
                    if(Mathf.Abs(theta2[k]) < ZERO_THRESH)
                    {
                        theta2[k] = 0;
                    }else if(theta2[k] < 0)
                    {
                        theta2[k] += 2 * Mathf.PI;
                    }
                    if(Mathf.Abs(theta4[k]) < ZERO_THRESH)
                    {
                        theta4[k] = 0;
                    }else if(theta4[k] < 0)
                    {
                        theta4[k] += 2 * Mathf.PI;
                    }
                    q_sols[num_sols * 6 + 0] = theta1[i];       q_sols[num_sols * 6 + 1] = theta2[k];
                    q_sols[num_sols * 6 + 2] = theta3[k];       q_sols[num_sols * 6 + 3] = theta4[k];
                    q_sols[num_sols * 6 + 4] = theta5[i][j];    q_sols[num_sols * 6 + 5] = theta6;
                    num_sols++;
                }`

And then I call the function in the start function to extract the needed joint angles from the position, then use them in Forward Kinematics.

` public void Start()
    {
        float[] q_sols = new float[8 * 6];
        float num_sols;
        slerpTime = 0;
        updateTargetRotations();
        num_sols =  VirtualInverse(TargetObject, q_sols);
        for (int i = 0; i < num_sols; i++) {
            print("%1.6f %1.6f %1.6f %1.6f %1.6f %1.6f\n");
            print(q_sols[i * 6 + 0] * 180 / Mathf.PI);
            print(q_sols[i * 6 + 1] * 180 / Mathf.PI);
            print(q_sols[i * 6 + 2] * 180 / Mathf.PI);
            print(q_sols[i * 6 + 3] * 180 / Mathf.PI);
            print(q_sols[i * 6 + 4] * 180 / Mathf.PI);
            print(q_sols[i * 6 + 5] * 180 / Mathf.PI);
        }
        for (int i = 0; i <= 4; i++)
        {
            print("%f ");
            print(Mathf.PI / 2.0 * i);
        }

    }`

gavanderhoorn, 2018-08-13:
I've not used Unity in this way nor have I looked at your code in detail.

Perhaps someone else monitoring this tracker can provide some assistance.

But posting this to a more suitable support channel -- such as ROS Answers, a Unity support forum or a general robotics forum -- might be better.

> I get the angle options but they are mega wrong

For anyone to possibly help you it would probably be good to quantify "mega wrong" a bit.

gavanderhoorn, 2018-08-13:
I'm actually going to close this, as this is not an issue with the packages in this repository per se.

If you find the kinematics are actually incorrect, and that is the cause of why your unity results don't work out -- or there is something else wrong with any of the packages in this repository then please feel free to re-open.

closed 2018-08-13 by gavanderhoorn