ros-industrial/universal_robot #381: Suggestion: rename repository

gavanderhoorn, 2018-08-27:
Simple really (or maybe not so much): rename the repository and metapackage to `universal_robots` from what it is now (`universal_robot`).

It's probably lost to history why this package and repository were created as `universal_robot`, but it's something that's been like that ever since the first version of the driver and description packages was created (eaff5f37).

ipa-nhg, 2019-02-07:
I am agree with you for the metapackage is quite trivial: https://github.com/ros-industrial/universal_robot/pull/401. Although til we release once the repo with the new name it doesn't take "real" effect.
The repository (and its url) is more sensitive for those that already have a fork and/or a local git clone, but communicating the change in the right channels should be enough.

Note: the wiki site has also to be renamed http://wiki.ros.org/universal_robot

davetcoleman, 2019-03-06:
renaming a package on github isn't a big deal because the old name will auto-redirect to the new name, even during ``git clone``

+1

gavanderhoorn, 2019-03-07:
This will have to be done tick-tock, in order to not break existing installation instructions.

davetcoleman, 2019-03-08:
> When you rename a repository, all existing information, with the exception of Project Pages URLs, is automatically redirected to the new name
> In addition to redirecting web traffic, all git clone, git fetch, or git push operations targeting the previous location will continue to function as if made on the new location.

gavanderhoorn, 2019-03-08:
Thanks @davetcoleman, but I'm well aware of how GH handles repository renames.

My comment was about more than just the name of the repository itself.

fmauch, 2022-11-07:
The meta-package itself is renamed already, the ROS [Wiki page](https://wiki.ros.org/universal_robots) is renamed and a redirection is in place. So, everything left to do should be actually renaming the repository and afterwards update the docs:

* [README](https://github.com/ros-industrial/universal_robot/blob/melodic-devel-staging/README.md)
* [ROS Wiki](https://wiki.ros.org/universal_robots)
* [ur_robot_driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#building)
* [ur_tutorials](https://github.com/UniversalRobots/Universal_Robots_ROS_Tutorials/blob/main/ur_documentation/docs/source/examples/installation.rst)

fmauch, 2022-11-10:
Removing this from the milestone, as this is not effectively stopping us from releasing.