ros-industrial/universal_robot #382: how to get urdf file？

lyhbuaa, 2018-08-30:
I need a ur10.urdf file， but there is none under the “universal_robot/ur_description/urdf/” folder，
so I used “rosrun xacro xacro.py ur10.urdf.xacro > ur10.urdf” to covert， but the generated ur10.urdf file is almost empty. Is there anyone kown why， or how to get the urdf file？

ipa-nhg, 2018-08-30:
@lyhbuaa  I think you are looking for:

`rosrun xacro xacro.py ur10_robot.urdf.xacro > ur10.urdf`

this will add a world link as reference, if you don't need this frame just create a new urdf.xacro file with the following content:

```
<?xml version="1.0"?>
<robot xmlns:xacro="http://ros.org/wiki/xacro" name="ur10" >

  <!-- ur10 -->
  <xacro:include filename="$(find ur_description)/urdf/ur10.urdf.xacro" />

  <!-- arm -->
  <xacro:ur10_robot prefix="" joint_limited="false"/>

</robot>
```
and run xacro to convert it to urdf.

gavanderhoorn, 2018-08-30:
@lyhbuaa: @ipa-nhg's comment should solve your problem.

As such I'll close this issue. You can still comment on it though, so feel free to do so if something is still not clear.

closed 2018-08-30 by gavanderhoorn

lyhbuaa, 2018-09-05:
@ipa-nhg ok，thank you very much, rosrun xacro xacro.py ur10_robot.urdf.xacro > ur10.urdf is all right!

lyhbuaa, 2018-09-05:
@gavanderhoorn ok!