ros-industrial/universal_robot #383: e-series robot models

adolfo-rt, 2018-09-03:
Is there any ongoing effort to support the e-series robots in this repo? The names of the branches seem to indicate that it's not the case, but I wanted to check if somebody has a local branch that I could contribute to / help to test.

gavanderhoorn, 2018-09-03:
Please see https://github.com/ThomasTimm/ur_modern_driver/issues/184.

So technically the any support in a driver will not be part of *this* repository, but the e-series models are contributed in #380.

> The names of the branches seem to indicate that it's not the case

I'm not sure I understand this comment.

adolfo-rt, 2018-09-03:
Thanks @gavanderhoorn. Will move any comments to the PR discussion. Closing here.

closed 2018-09-03 by adolfo-rt

gavanderhoorn, 2018-09-03:
Note: #380 only adds urdfs and moveit configs, it doesn't include any driver development or changes.

adolfo-rt, 2018-09-03:
Fortunately, it's as much as I need at the moment to move forward.