ros-industrial/universal_robot #384: UR5 arm is protective stopped during execution without touching any obstacles

shinian123, 2018-09-05:
Our lab use ROS operating system to operate UR5 arms using moveit. Recently,we encounterd a problem when the UR5 arms get protective stop during execution. While the arm does not touch any obstacles.

Our joint_limits.yaml is : 
joint_limits:
  left_arm_elbow_joint:
    has_velocity_limits: true
    max_velocity: 3.15
    has_acceleration_limits: false
    max_acceleration: 0
  left_arm_shoulder_lift_joint:
    has_velocity_limits: true
    max_velocity: 3.15
    has_acceleration_limits: false
    max_acceleration: 0
  left_arm_shoulder_pan_joint:
    has_velocity_limits: true
    max_velocity: 3.15
    has_acceleration_limits: false
    max_acceleration: 0
  left_arm_wrist_1_joint:
    has_velocity_limits: true
    max_velocity: 3.2
    has_acceleration_limits: false
    max_acceleration: 0
  left_arm_wrist_2_joint:
    has_velocity_limits: true
    max_velocity: 3.2
    has_acceleration_limits: false
    max_acceleration: 0
  left_arm_wrist_3_joint:
    has_velocity_limits: true
    max_velocity: 3.2
    has_acceleration_limits: false
    max_acceleration: 0
  right_arm_elbow_joint:
    has_velocity_limits: true
    max_velocity: 3.15
    has_acceleration_limits: false
    max_acceleration: 0
  right_arm_shoulder_lift_joint:
    has_velocity_limits: true
    max_velocity: 3.15
    has_acceleration_limits: false
    max_acceleration: 0
  right_arm_shoulder_pan_joint:
    has_velocity_limits: true
    max_velocity: 3.15
    has_acceleration_limits: false
    max_acceleration: 0
  right_arm_wrist_1_joint:
    has_velocity_limits: true
    max_velocity: 3.2
    has_acceleration_limits: false
    max_acceleration: 0
  right_arm_wrist_2_joint:
    has_velocity_limits: true
    max_velocity: 3.2
    has_acceleration_limits: false
    max_acceleration: 0
  right_arm_wrist_3_joint:
    has_velocity_limits: true
    max_velocity: 3.2
    has_acceleration_limits: false
    max_acceleration: 0

It seems when the speed of the arms changed drastically , the protective stop program will be triggered.
We think the problem is due to some parameters of joint_limits.yaml is not correctly set, but we don't know how to tune these parameters.Could any one help us? Thanks.

atyshka, 2018-10-27:
Yeah same problem here. All I can say is set the acceleration to low (but not zero) and the velocity to under 1.0

gstavrinos, 2019-01-23:
We are experiencing the same behaviour with our UR3. I have used the same method described by @atyshka , but we still have some protective stops, but very rarely. I might have not lowered the acceleration and velocity enough though.

Just a heads-up to bring some attention to this issue.

atyshka, 2019-01-23:
@gstavrinos Are you using a wired or wireless connection? Latency matters. Also, you can change the safety settings on the UR control box to set the limits. We set safety to the maximum setting (high velocity and acceleration limits) and noticed improved results. You could also try modifying the payload setting, if you have a heavier payload, the force of accelerating it may cause the UR to think it something.

gstavrinos, 2019-01-24:
We are using a wired connection and I have tried adjusting the payload setting, but I haven't played with the safety settings at all, so I will definitely give them a shot! Thanks for the help, @atyshka!

nazir-ust, 2019-07-23:
Any luck resolving this issue? I'm also facing it with our UR10

gstavrinos, 2019-07-24:
@nazir-ust Playing with the payload and safety settings definitely helps, but I still experience this issue from time to time. It's not that regular, at least.

nazir-ust, 2019-07-24:
Thank you. Good to know. I was able to somewhat address by tweaking with the waypoints array. Wasn't creating it accurately before and there were large fluctuations. Perhaps Moveit time parameterizes based on individual separation of waypoints.

gavanderhoorn, 2019-11-23:
I'm not sure which driver was used by the OP, but I'm closing this as with the deprecation of `ur_modern_driver` we'll also remove `ur_driver` from this repository in `melodic-devel` (#448).

Refer to the [announcement on ROS Discourse](https://discourse.ros.org/t/deprecation-of-ur-modern-driver/11629).

`ur_robot_driver` uses a different approach to all of this, so it's unclear whether the issue reported in the OP is still actual.

If it is, it should be re-opened on the `ur_robot_driver` issue tracker.

closed 2019-11-23 by gavanderhoorn