ros-industrial/universal_robot #386: UR10 Gazebo Launching Problem

ipa-mjp, 2018-10-11:
Hello all,
I am using UR10 arm. While lunching gazebo model of UR10 arm in the kinetic, it is falling down on the ground. I think but not sure this is the problem of the gravity. I tried to tune this parameter but not successed yet. If any one facing same problem and solve it than please let me know.

My setup:

- Ubuntu 16.04
- ROS Kinetic
- UR10 arm

shloakmehta, 2018-11-16:
So I get these errors too which - according to the last two warnings - gravity acts weird, and to solve it, we either have to  set gazebo_pid parameters, switch to the VelocityJointInterface or EffortJointInterface, or upgrade to Gazebo 9.

I may be wrong and am about to try myself right now with the UR10 but according to https://github.com/ros-simulation/gazebo_ros_pkgs/issues/612 it should be the solution. 

Let me know how you get on...
```
[ERROR] [1542388350.763274103, 0.342000000]: No pgain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/shoulder_pan_joint
[ERROR] [1542388350.765508662, 0.342000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/shoulder_lift_joint
[ERROR] [1542388350.767169872, 0.342000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/elbow_joint
[ERROR] [1542388350.768819330, 0.342000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_1_joint
[ERROR] [1542388350.770428920, 0.342000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_2_joint
[ERROR] [1542388350.772307131, 0.342000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_3_joint
[ INFO] [1542388350.785158214, 0.342000000]: Loaded gazebo_ros_control.
[ WARN] [1542388350.786564225, 0.343000000]: The default_robot_hw_sim plugin is using the Joint::SetPosition method without preserving the link velocity.
[ WARN] [1542388350.786622491, 0.343000000]: As a result, gravity will not be simulated correctly for your model.
[ WARN] [1542388350.786664522, 0.343000000]: Please set gazebo_pid parameters, switch to the VelocityJointInterface or EffortJointInterface, or upgrade to Gazebo 9.

```

ipa-nhg, 2019-02-07:
@ipa-mjp could you please give some details about how did you started the simulation?

This command `roslaunch ur_gazebo ur10.launch` ?? For kinetic and gazebo 7 this should work. Usually the error you have (as @shloakmehta pointed) is a problem with the controllers, the log should point to the concrete problem. Generally, an update of the controllers is strongly recommended for this cases. Also check that all the required packages are installed:

`rosdep install --from-path src/universal_robot/ur_gazebo`

@shloakmehta I have never had this issue, unfortunately I can't reproduce it.  did you solve the problem? Are you mounting somewhere the arm or is  it a "arm-solo setup"?

ChanWaiYang, 2019-02-16:
@ipa-mjp Did you run this command line 
> sudo apt-get install ros-kinetic-universal-robot

padhupradheep, 2019-03-21:
Hey,

ROS Distro: Kinetic
Gazebo Version: 7.0
I'm also facing the same issue with the position controller. If I use effort controllers or velocity controller as per the https://github.com/ros-simulation/gazebo_ros_pkgs/issues/612 , I'm not sure on what PID values that needs to be set for the gains of both the arm controller and the gazebo ros controller.  As I tried to follow this discussion in https://answers.ros.org/question/284241/mobile-arm-simulation-behaves-strangely-when-adding-ros_control-plugin-for-arm/ , still the robot wobbles. 

Any help will be appreciated!

ipa-nhg, 2019-07-05:
Unfortunately set the `pid` gains just break the arm and can't be controlled anymore. Our current version is working, although Gazebo shows the errors. Then I just can suggest to ignore the errors.
I open a PR with my experiments, keeping the `pid` configuration commented out for now:
https://github.com/ros-industrial/universal_robot/pull/443

cbw36, 2020-07-07:
Does this still need to be developed for kinetic? I have had issues in the past with UR in gazebo where I cannot actuate the joints using kinetic, gazebo7 and 16.04

gavanderhoorn, 2020-07-07:
There has been quite some work on all packages in this repository in the context of #448.

I've recently updated the Gazebo packages myself and I've not encountered any issues.

We will not be backporting any of that to Kinetic.

gavanderhoorn, 2020-07-07:
Please see #443 as well.