ros-industrial/universal_robot #390: [moveit, simulation] "Failed to fetch current robot state" when running getCurrentPose

umhan35, 2018-11-02:
I want to get the pose of the UR5 end effecor using moveit, but moveit complains failure to get the current robot state.

moveit did subscribe to /joint_states (one subscriber is /move_group)

```
$ rostopic info /joint_states
Type: sensor_msgs/JointState

Publishers: 
 * /gazebo (http://mass:42821/)

Subscribers: 
 * /robot_state_publisher (http://mass:45219/)
 * /move_group (http://mass:40581/)
 * /a2 (http://mass:37295/)
```

Here is the full story:

I am running ROS Kinetic on Ubuntu 16.04

ur5 is in simulation:

```
//Gazebo:
roslaunch ur_gazebo ur5.launch

//Rviz
roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true
roslaunch ur5_moveit_config moveit_rviz.launch config:=true
```

Here is the code:

```
#include <ros/ros.h>
#include <moveit/move_group_interface/move_group_interface.h>

int main(int argc, char **argv) {

    ros::init(argc, argv, "a2");

    ros::NodeHandle node;

    moveit::planning_interface::MoveGroupInterface ur5("manipulator");

    std::cout << ur5.getCurrentPose();

    ros::spin();

    return EXIT_SUCCESS;
}
```

Here is the output.

```
[ INFO] [1541172002.952745654]: Loading robot model 'ur5'...
[ WARN] [1541172002.952787544]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1541172002.952799465]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1541172002.989670179]: Loading robot model 'ur5'...
[ WARN] [1541172002.989693771]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1541172002.989705695]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1541172003.946774896, 4402.149000000]: 
Deprecation warning: Trajectory execution service is deprecated (was replaced by an action).
Replace 'MoveGroupExecuteService' with 'MoveGroupExecuteTrajectoryAction' in move_group.launch
[ INFO] [1541172003.948808649, 4402.151000000]: Ready to take commands for planning group manipulator.
[ INFO] [1541172004.950926256, 4403.151000000]: Didn't received robot state (joint angles) with recent timestamp within 1 seconds.
Check clock synchronization if your are running ROS across multiple machines!
header: 
  seq: 0
[ERROR] [1541172004.950960139, 4403.151000000]: Failed to fetch current robot state
  stamp: 4403.151000000
  frame_id: /world
pose: 
  position: 
    x: 0
    y: 0
    z: 0
  orientation: 
    x: 0
    y: 0
    z: 0
    w: 1

```

Thanks in advance!

gavanderhoorn, 2018-11-02:
Can you check whether `gazebo_ros_control` publishes `JointState`s on `/joint_states` using `rostopic echo -n1 /joint_states`?

If it does, then this is not an issue with the package(s) in this repository, and you should look at the MoveIt side and/or your code.

gavanderhoorn, 2018-11-02:
I'm going to close this assuming that the simulator does in fact publish `JointState` messages.

If it doesn't, then please re-open and we can try to figure out why it isn't doing that.

Regardless: feel free to keep commenting on this issue of course.

closed 2018-11-02 by gavanderhoorn

umhan35, 2018-11-04:
It does publish on `/joint_states`. I will post this issue over moveit but I really hope someone here could help because this issue also involves the moveit config.