ros-industrial/universal_robot #397: ur_gazebo No p gain specified for pid

dovanhuong, 2019-01-14:
Dear all,
 I'm using gazebo 7 and ur5. I'm trying to launch ur_gazebo ur5.launch. I updated the latest universal robot file in the link: https://github.com/ros-industrial/universal_robot
 It's always tell to me that: <p style="color:red;">No p gain specified for pid</p> 
Even, I tried to work with older version, it is the same issue like that!
This is the detail:

[ INFO] [1547472445.585903463, 0.083000000]: Starting gazebo_ros_control plugin in namespace: /
[ INFO] [1547472445.587197301, 0.083000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [robot_description] on the ROS param server.
[ERROR] [1547472445.717618844, 0.083000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/shoulder_pan_joint
[ERROR] [1547472445.721082032, 0.083000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/shoulder_lift_joint
[ERROR] [1547472445.723013939, 0.083000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/elbow_joint
[ERROR] [1547472445.725118694, 0.083000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_1_joint
[ERROR] [1547472445.728197118, 0.083000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_2_joint
[ERROR] [1547472445.731720743, 0.083000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_3_joint
[ INFO] [1547472445.742479460, 0.083000000]: Loaded gazebo_ros_control.

If anyone know the root cause, please tell me know! 
Thanks you in advance,

dovanhuong, 2019-01-15:
Dear everyone,

I'd send you my full description of my issue and link of package I'm using as below: 
https://github.com/dovanhuong/universal_robot_issue_ur_gazebo_ur5.launch
Issue description: 
huong@huong:/catkin_ws$ source devel/setup.bash
huong@huong:/catkin_ws$ roslaunch ur_gazebo ur5.launch
... logging to /home/huong/.ros/log/1fbc3f3e-18bc-11e9-8bfd-bc5ff4c9bc1b/roslaunch-huong-13148.log Checking log directory for disk usage. This may take awhile. Press Ctrl-C to interrupt Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://huong:44767/
SUMMARY

PARAMETERS

    /arm_controller/action_monitor_rate: 10
    /arm_controller/constraints/elbow_joint/goal: 0.1
    /arm_controller/constraints/elbow_joint/trajectory: 0.1
    /arm_controller/constraints/goal_time: 0.6
    /arm_controller/constraints/shoulder_lift_joint/goal: 0.1
    /arm_controller/constraints/shoulder_lift_joint/trajectory: 0.1
    /arm_controller/constraints/shoulder_pan_joint/goal: 0.1
    /arm_controller/constraints/shoulder_pan_joint/trajectory: 0.1
    /arm_controller/constraints/stopped_velocity_tolerance: 0.05
    /arm_controller/constraints/wrist_1_joint/goal: 0.1
    /arm_controller/constraints/wrist_1_joint/trajectory: 0.1
    /arm_controller/constraints/wrist_2_joint/goal: 0.1
    /arm_controller/constraints/wrist_2_joint/trajectory: 0.1
    /arm_controller/constraints/wrist_3_joint/goal: 0.1
    /arm_controller/constraints/wrist_3_joint/trajectory: 0.1
    /arm_controller/gains/elbow_joint/d: 0.1
    /arm_controller/gains/elbow_joint/i: 0.0
    /arm_controller/gains/elbow_joint/i_clamp: 1
    /arm_controller/gains/elbow_joint/p: 100.0
    /arm_controller/gains/shoulder_lift_joint/d: 0.1
    /arm_controller/gains/shoulder_lift_joint/i: 0.0
    /arm_controller/gains/shoulder_lift_joint/i_clamp: 1
    /arm_controller/gains/shoulder_lift_joint/p: 100.0
    /arm_controller/gains/shoulder_pan_joint/d: 0.1
    /arm_controller/gains/shoulder_pan_joint/i: 0.0
    /arm_controller/gains/shoulder_pan_joint/i_clamp: 1
    /arm_controller/gains/shoulder_pan_joint/p: 100.0
    /arm_controller/gains/wrist_1_joint/d: 0.1
    /arm_controller/gains/wrist_1_joint/i: 0.0
    /arm_controller/gains/wrist_1_joint/i_clamp: 1
    /arm_controller/gains/wrist_1_joint/p: 100.0
    /arm_controller/gains/wrist_2_joint/d: 0.1
    /arm_controller/gains/wrist_2_joint/i: 0.0
    /arm_controller/gains/wrist_2_joint/i_clamp: 1
    /arm_controller/gains/wrist_2_joint/p: 100.0
    /arm_controller/gains/wrist_3_joint/d: 0.1
    /arm_controller/gains/wrist_3_joint/i: 0.0
    /arm_controller/gains/wrist_3_joint/i_clamp: 1
    /arm_controller/gains/wrist_3_joint/p: 100.0
    /arm_controller/joints: ['shoulder_pan_jo...
    /arm_controller/state_publish_rate: 25
    /arm_controller/stop_trajectory_duration: 0.5
    /arm_controller/type: velocity_controll...
    /joint_state_controller_ur_gazebo/publish_rate: 50
    /joint_state_controller_ur_gazebo/type: joint_state_contr...
    /robot_description: <?xml version="1....
    /robot_state_publisher/publish_frequency: 50.0
    /robot_state_publisher/tf_prefix:
    /rosdistro: kinetic
    /rosversion: 1.12.14
    /use_sim_time: True

NODES / arm_controller_spawner (controller_manager/controller_manager) fake_joint_calibration (rostopic/rostopic) gazebo (gazebo_ros/gzserver) gazebo_gui (gazebo_ros/gzclient) joint_state_controller_spawner (controller_manager/controller_manager) robot_state_publisher (robot_state_publisher/robot_state_publisher) spawn_gazebo_model (gazebo_ros/spawn_model)

auto-starting new master process[master]: started with pid [13163] ROS_MASTER_URI=http://localhost:11311

setting /run_id to 1fbc3f3e-18bc-11e9-8bfd-bc5ff4c9bc1b process[rosout-1]: started with pid [13177] started core service [/rosout] process[gazebo-2]: started with pid [13201] process[gazebo_gui-3]: started with pid [13206] process[spawn_gazebo_model-4]: started with pid [13211] process[robot_state_publisher-5]: started with pid [13212] process[fake_joint_calibration-6]: started with pid [13213] process[joint_state_controller_spawner-7]: started with pid [13214] process[arm_controller_spawner-8]: started with pid [13223] [ INFO] [1547553188.745573927]: Finished loading Gazebo ROS API Plugin. [ INFO] [1547553188.745766763]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting... [ INFO] [1547553188.761690892]: Finished loading Gazebo ROS API Plugin. [ INFO] [1547553188.761994532]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting... SpawnModel script started [INFO] [1547553189.074498, 0.000000]: Loading model XML from ros parameter [INFO] [1547553189.076750, 0.000000]: Waiting for service /gazebo/spawn_urdf_model [INFO] [1547553189.681114, 0.000000]: Calling service /gazebo/spawn_urdf_model [INFO] [1547553189.870412, 0.000000]: Spawn status: SpawnModel: Successfully spawned entity [ INFO] [1547553189.894951360]: Loading gazebo_ros_control plugin [ERROR] [1547553189.895135241]: GazeboRosControlPlugin missing while using DefaultRobotHWSim, defaults to true. This setting assumes you have an old package with an old implementation of DefaultRobotHWSim, where the robotNamespace is disregarded and absolute paths are used instead. If you do not want to fix this issue in an old package just set to true.

[... cut ...] sh joint: wrist_3_joint [ INFO] [1547553190.027451016]: Starting GazeboRosJointStatePublisher Plugin (ns = robot/)!, parent name: robot [spawn_gazebo_model-4] process has finished cleanly log file: /home/huong/.ros/log/1fbc3f3e-18bc-11e9-8bfd-bc5ff4c9bc1b/spawn_gazebo_model-4*.log Loaded joint_state_controller_ur_gazebo 

<br><br>
I'm really looking forward to receive support from everyone in my case, I spent for a long time, unfortunately I was not fix this issue!. Thank you so much!
Sincerely,

sapan-ostic, 2019-03-24:
Adding following lines to .yaml file worked for me.

/gazebo_ros_control:
  pid_gains:
    shoulder_pan_joint:
      p: 100.0
      i: 0.01 
      d: 10.0
    shoulder_lift_joint:
      p: 100.0
      i: 0.01 
      d: 10.0  
    elbow_joint:
      p: 100.0
      i: 0.01 
      d: 10.0
    wrist_1_joint:
      p: 100.0
      i: 0.01 
      d: 10.0
    wrist_2_joint:
      p: 100.0
      i: 0.01 
      d: 10.0
    wrist_3_joint:
      p: 100.0
      i: 0.01 
      d: 10.0

janisa9, 2019-04-18:
> Adding following lines to .yaml file worked for me.
>
> /gazebo_ros_control:
> pid_gains:
> shoulder_pan_joint:
> p: 100.0
> i: 0.01
> d: 10.0
> shoulder_lift_joint:
> p: 100.0
> i: 0.01
> d: 10.0
> elbow_joint:
> p: 100.0 [...]

In which .yaml you added these lines?

davizinho5, 2019-05-31:
In my case, I'm working  with the universal packages in Ubuntu 18.04 and ROS-melodic.
I installed the packages from source and downloaded all the dependencies manually.

Adding this control parameters just made it worse, see #427 for my problems/solution.

ar-i-enterprise, 2020-05-05:
> Adding following lines to .yaml file worked for me.
>
> /gazebo_ros_control:
> pid_gains:
> shoulder_pan_joint:
> p: 100.0
> i: 0.01
> d: 10.0
> shoulder_lift_joint:
> p: 100.0
> i: 0.01
> d: 10.0
> elbow_joint:
> p: 100.0 [...]

which yaml and how you get there?

Spandre, 2020-05-07:
I have a similar problem, but in my case when adding the PID values for the arm it seems to be broken. I add two images with and without the PID controller in the .yaml file only for the arm (for the gripper the pid controller is specified and works well). Anyway when not specified, I have the error no p gain specified for PID.
[image]
[image]

KavitShah1998, 2020-09-28:
> Adding following lines to .yaml file worked for me.
>
> /gazebo_ros_control:
> pid_gains:
> shoulder_pan_joint:
> p: 100.0
> i: 0.01
> d: 10.0
> shoulder_lift_joint:
> p: 100.0
> i: 0.01
> d: 10.0
> elbow_joint:
> p: 100.0 [...]

Just adding into it.

Steven-ZY, 2020-11-24:
@KavitShah1998 Sir, thank you so much for your detailed reply. Well, according to your instructions, the issues about "pids" have been solved. However, a new problem has arisen: when I `roslaunch xxx`, the robot's posture changed, that's not what it should be, that's to say it's posture is not original. I'm looking forward to your reply. Thanks in advance.

KavitShah1998, 2020-11-24:
@Steven-ZY You can solve the issue about robot pose by specifying the pose in the launch file for your robot.
You can do so by 
1. Set pose in roslaunch [Link](https://answers.ros.org/question/339163/change-turtlebot-initial-position-in-gazebo-ros/) 
2. Set ROBOT_INITIAL_POSE variable through your terminal or even .bashrc . [Link]{https://answers.ros.org/question/341026/is-there-a-way-to-specify-the-initial-position-of-the-robotic-from-world-file/}
3. Set robot pose in urdf file of the robot (Not preferred)

For more questions on this issue, I would recommend you to create a new issue for the same