ros-industrial/universal_robot #399: Display the joint angle limits in RViz/Gazebo

moucrob, 2019-02-06:
Hello,

Usually to feel the angle limits of the robots I work on, I add a tf plugin in RViz.
However for centrally symmetric links such as UR ones (kukas or frankas are a bit weirder), it may sometimes be difficult to sense. 

Would it exist a way to rapidly draw a line along the cylindrical [link meshes `.dae`](https://github.com/ros-industrial/universal_robot/tree/kinetic-devel/ur_description/meshes/ur10/visual) ? Does a mesh design frame always matches its associated tf frame in RViz ? (in term of rotation at least)

### People to bother
@fmessmer @gavanderhoorn  @shaun-edwards @kphawkins  according to [here](http://wiki.ros.org/universal_robot), sorry guys

gavanderhoorn, 2019-02-06:
> Would it exist a way to rapidly draw a line along the cylindrical [link meshes `.dae`](https://github.com/ros-industrial/universal_robot/tree/kinetic-devel/ur_description/meshes/ur10/visual) ?

I don't know of such a way. At least no in RViz.

It might be better to ask this on ROS Answers, as there's many more people that could potentially answer your question there.

> Does a mesh design frame always matches its associated tf frame in RViz ? (in term of rotation at least)

I'm not sure I completely understand what you're asking, but if you're asking whether the origin of meshes used in robot description / support pkgs always matches the TF frame, then the answer would be "no". URDF supports arbitrary orientations of meshes relative to the orientation of the `link` (and even multiple meshes per `link`).

> @fmessmer @gavanderhoorn  @shaun-edwards @kphawkins  according to [here](http://wiki.ros.org/universal_robot), sorry guys

Please don't do this. Whoever is currently in charge of maintaining this repository will know. There is no point in bothering historical contributors (and it's also somewhat annoying actually).

gavanderhoorn, 2019-02-06:
I'm going to close this as it's not an issue with the packages in this repository.

Feel free to keep commenting on it though.

closed 2019-02-06 by gavanderhoorn