ros-industrial/universal_robot #400: Joints locked in origin in Rviz

Namrell, 2019-02-06:
Hi,
Backgroun: I am trying to use the github Repo for the Universal Robotis with ROS, MoveIt and Gazebo, however the arm is flailing around when I try to move in a small increment, so I tried to open it in Rviz in order to see if I could move the arm there. However in Rviz the UR-robot seems to have all it's joints locked in the origin (see the attached picture). When I try to make the arm move around its moving with all it's joints around the origin. I am guessing this could be something with how the robot arm is defined for Rviz/MoveIt. I am using ROS melodic, Gazebo 9.0.0, RVIZ 1.13.1.

As far as I can see I get no error messages. Do you have any ideas what could be causing this?

[image]

gavanderhoorn, 2019-02-06:
Please see #374, most likely this is a locale issue.

Namrell, 2019-02-06:
You are right. A shame my search-engine skills did not manage to find that issue earlier :). Thanks!

closed 2019-02-06 by Namrell