ros-industrial/universal_robot #411: limit to amount of targets

Chriszlf, 2019-03-21:
I'm now trying to send 5000 targets to a Bartlett UR10 robot via ROS. But the robot gives no reaction at all. It looks like there is a limit to the amount of targets. Could anyone give some suggestion on such a case?

gavanderhoorn, 2019-03-21:
We're going to need (a lot) more information before we can say anything about this.

What setup are you using? How exactly are you sending "targets"?

Which software are you using? Which topics/services/actions?

Etc etc

Chriszlf, 2019-03-21:
Hi,
we are sending a UR script via the ur_modern driver on the topic /ur_driver/URScript
Less than 1.000 targets can be sent and executed, more than 1.000 don't work.
We are using a Plugin for our CAD Software Rhino 3D that communicates via ROS Bridge.
We also tried to upload the code directly to the teaching pendant via USB stick. Though it crushed as well and gives the warning: 

> infinite loop

 

`
def Program():
  lfTcp = p[0.00000, 0.00000, 0.00000, 0.0000, 0.0000, -1.5708]
  lfWeight = 1.000
  lfCog = [0.00000, 0.00000, 0.00000]
  Speed000 = 0.25000
  Speed001 = 0.01500
  DefaultZone = 0.00000
  Zone000 = 0.00350
  set_tcp(lfTcp)
  set_payload(lfWeight, lfCog)
  movej([-0.8425, -2.2402, -1.7640, -2.2790, 0.7283, -1.5708], a=3.1416, v= 0.314, r=DefaultZone)
  movel(p[0.15750, -0.80875, -0.21170, -1.2092, -1.2092, 1.2092],a=1.00000,v=Speed001,r=Zone000)
  movel(p[0.15749, -0.81282, -0.21667, -1.2092, -1.2092, 1.2092],a=1.00000,v=Speed001,r=Zone000)
  movel(p[0.15748, -0.81800, -0.22047, -1.2092, -1.2092, 1.2092],a=1.00000,v=Speed001,r=Zone000)
  .
  .
  .
  movel(p[0.09756, -0.88425, -0.15745, -1.2092, -1.2092, 1.2092],a=1.00000,v=Speed001,r=Zone000)
  movel(p[0.09750, -0.88465, -0.15815, -1.2092, -1.2092, 1.2092],a=1.00000,v=Speed001,r=Zone000)
  movej([-0.8425, -2.2402, -1.7640, -2.2790, 0.7283, -1.5708], a=3.1416, v= 0.438, r=DefaultZone)
end

`
Is it possible that the UR controller cant deal with more than 1.000 targets?

Thanks!

gavanderhoorn, 2019-03-21:
> Is it possible that the UR controller cant deal with more than 1.000 targets?

could very well be.

But that would be something to ask UR.

gavanderhoorn, 2019-03-21:
For now I'm going to assume this is not a problem with the packages in this repository and as such I'll close this issue.

If you feel that is incorrect, or have more information please re-open and we can try to diagnose further.

closed 2019-03-21 by gavanderhoorn