ros-industrial/universal_robot #412: Getting Error [Err] [REST.cc:205] Error in REST request

sapan-ostic, 2019-03-24:
Hi, 
I am using ROS Melodic.

I get an error:
[Err] [REST.cc:205] Error in REST request

libcurl: (51) SSL: no alternative certificate subject name matches target host name 'api.ignitionfuel.org'

Also the model looks like:

[image]

gavanderhoorn, 2019-03-24:
This has nothing to do with `universal_robot`, but is most likely a Gazebo error (seeing as the Gazebo `api.ignitionfuel.org` host is mentioned).

I would suggest to ask for support over at Gazebo Answers or perhaps even the Gazebo issue tracker.

gavanderhoorn, 2019-03-24:
I'm closing this (as it's most likely not an issue with the packages in this repository), but if it does turn out to be something we can do something about, please re-open.

Feel free to keep commenting on the issue of course.

closed 2019-03-24 by gavanderhoorn

JonathanPlasse, 2019-07-30:
Hi,
Inside `~/.ignition/fuel/config.yaml` replace

`url: https://api.ignitionfuel.org`
by
`url: https://api.ignitionrobotics.org`

It solved the problem for me.

gavanderhoorn, 2019-08-01:
It's so nice to see Gazebo/Ignition devs/maintainers keeping redirects in place for bw-compatibility ..