ros-industrial/universal_robot #423: Complex numbers of q5

dwadelinjianxiong, 2019-05-09:
double q5[2][2];
    {
      for(int i=0;i<2;i++) {
        double numer = (T03*sin(q1[i]) - T13*cos(q1[i])-d4);
        double div;
        if(fabs(fabs(numer) - fabs(d6)) < ZERO_THRESH)
          div = SIGN(numer) * SIGN(d6);
        else
          div = numer / d6;
        double arccos = acos(div);
        q5[i][0] = arccos;
        q5[i][1] = 2.0*PI - arccos;
      }
    }
     Here, I noticed that when you are calculating q5, not discussing about the situation that numer>d6 to make function acos() return complex numbers, I want to know that is there possible of my consideration.

dwadelinjianxiong, 2019-05-12:
[image]

dwadelinjianxiong, 2019-05-12:
[image]
In this paper, it discussed the situation that numer>d6 which your code not, I was wondering if this a possible situation make acos() return plural since div = numer / d6 is greater than 1. Then the q5 will not make sense.

gavanderhoorn, 2019-05-12:
While I haven't checked the math in the code, I can say that `acos(..)` as used in the kinematics plugin cannot return complex numbers. From [the man page](https://linux.die.net/man/3/acos):

```Roff
ACOS(3)                                                                    Linux Programmer's Manual                                                                   ACOS(3)

NAME
       acos, acosf, acosl - arc cosine function

SYNOPSIS
       #include <math.h>

       double acos(double x);
       float acosf(float x);
       long double acosl(long double x);

       Link with -lm.

   Feature Test Macro Requirements for glibc (see feature_test_macros(7)):

       acosf(), acosl():
           _BSD_SOURCE || _SVID_SOURCE || _XOPEN_SOURCE >= 600 || _ISOC99_SOURCE || _POSIX_C_SOURCE >= 200112L;
           or cc -std=c99

DESCRIPTION
       These functions calculate the arc cosine of x; that is the value whose cosine is x.

RETURN VALUE
       On success, these functions return the arc cosine of x in radians; the return value is in the range [0, pi].

       If x is a NaN, a NaN is returned.

       If x is +1, +0 is returned.

       If x is positive infinity or negative infinity, a domain error occurs, and a NaN is returned.

       If x is outside the range [-1, 1], a domain error occurs, and a NaN is returned.

ERRORS
       See math_error(7) for information on how to determine whether an error has occurred when calling these functions.

       The following errors can occur:

       Domain error: x is outside the range [-1, 1]
              errno is set to EDOM.  An invalid floating-point exception (FE_INVALID) is raised.
```

dwadelinjianxiong, 2019-05-12:
```
double q1[2];
    {
      ...
      else if(d4*d4 > R) {
        return num_sols;
      }
      else {
        double arccos = acos(d4 / sqrt(R)) ;
        ...
      }
    }
```

Here, eliminate the situation d4*d4 > R to avoid d4 / sqrt(R) being greater than 1;

```
double q2[2], q3[2], q4[2];
          ///////////////////////////// RRR joints (q2,q3,q4) ////////////////////////////
          ...
          else if(fabs(c3) > 1.0) {
            // TODO NO SOLUTION
            continue;
          }
          double arccos = acos(c3);
         ...
```
Here, eliminate the situation fabs(c3) > 1.0 to avoid c3 being greater than1; 
Then I was confused by the following code that why you are not eliminate the situation that div = numer / d6 is greater than 1 since it might return error NaN as you showed.
```
double q5[2][2];
    {
      for(int i=0;i<2;i++) {
        double numer = (T03*sin(q1[i]) - T13*cos(q1[i])-d4);
        double div;
        if(fabs(fabs(numer) - fabs(d6)) < ZERO_THRESH)
          div = SIGN(numer) * SIGN(d6);
        else
          div = numer / d6;
        double arccos = acos(div);
        q5[i][0] = arccos;
        q5[i][1] = 2.0*PI - arccos;
      }
    }
```

kphawkins, 2019-05-23:
If q1 is solved for correctly, (7) should hold to machine precision:
[image]
If you solve for the argument in the cosine function you'll find it is equal to z_x s_1 - z_y c_1, and this expression should have magnitude at most 1.  Theoretically possible to exceed that value for strange values of d_4/d_6 and definitely possible if z_x/z_y are not proper values of an SO(3) matrix, but shouldn't happen in general.

dwadelinjianxiong, 2019-05-24:
got it, thanks!

closed 2019-05-24 by dwadelinjianxiong

gavanderhoorn, 2019-05-24:
Thanks @kphawkins for the insights. Appreciated.