ros-industrial/universal_robot #424: Custom end effector and MoveIt?

cschuwerk, 2019-05-22:
Hi,
quick question: is there any designated way to use to extend the available packages (ur_description, urx_moveit_config) with a customized end effector attached to the UR? In my case, it would be sufficient to have a fixed offset to the UR's tool frame which is considered in MoveIt for path planning and IK.

I guess this is some standard feature many people require, but I could not find any information on how to approach this. If such a feature is not available, the proper way to go is to create a new moveit config for the setup at hand, I guess!?

Thanks for helping.
Clemens

gavanderhoorn, 2019-05-22:
> I guess this is some standard feature many people require, but I could not find any information on how to approach this.

In general I don't recommend reusing moveit configurations, for the simple reason that running the wizard is a 5 minute job, you're sure that you get a configuration that is completely up-to-date for your specific robot setup and it will take all changes you made to your robot and environment into account.

Editing a package with auto-generated files is not recommended in any case, and I would also not recommend it here.

The same goes for description packages: the recommendation is to not touch any files in such packages, but create a new package in which you reference files in the base description pkg.

For adding EEFs or toolframes, you'd create a xacro macro specific to your application/robot setup, name it appropriate and as part of that macro, you'd first instantiate the base robot model, then add your customisations and finally link everything together with an appropriate joint (a `fixed` on in most cases).

Now run the MoveIt Setup Assistant on your new, *composite*, xacro and you have a MoveIt configuration specific to your application (which should probably also include a model of your environment).

concluding: 

> the proper way to go is to create a new moveit config for the setup at hand, I guess!?

yes, that would be the way to go about this, but not for the reason you imply.

gavanderhoorn, 2019-05-22:
As this is a question about usage and configuration, not an actual actionable issue, I'm going to close this.

Feel free to keep commenting on the issue of course, @cschuwerk.

closed 2019-05-22 by gavanderhoorn

cschuwerk, 2019-05-22:
Thx for your fast reply and help. Extending the available UR macro with my own macro in a separate file was exactly my plan.

gavanderhoorn, 2019-05-22:
Just to be extra clear: you wouldn't be extending it. You would be composing your own xacro, in which you make use of the provided xacro macros.