ros-industrial/universal_robot #425: Bad quality visual meshes for UR3 in ros Kinetic

cocodmdr, 2019-05-29:
[image]

Visual meshes are bad quality for UR3 in ros kinetic especially the wrist3.dae. 

The mesh is not even centered when moving it with the joint_state_puplisher.

[image]

gavanderhoorn, 2019-05-29:
That does indeed not look very good.

Is that the regular UR3 model, or the e-series one?

It might be good to check whether that has always been that way or if things have deteriorated due to post-processing in a contributed PR.

History:

 - [ur_description](https://github.com/ros-industrial/universal_robot/commits/kinetic-devel/ur_description)
 - [ur_e_description](https://github.com/ros-industrial/universal_robot/commits/kinetic-devel/ur_e_description)

cocodmdr, 2019-05-29:
It is the regular UR3, there are topics about bad shading in the history.  
[(https://github.com/ros-industrial/universal_robot/issues/233)]
However I believe the wrist3.stl was forgotten in this commit.

gavanderhoorn, 2019-05-29:
> However I believe the wrist3.dae was forgotten only for the UR3 in this commit by @j-polden : [da652cf#diff-ae6e1415a6620b312668ab17564ad6e0](https://github.com/ros-industrial/universal_robot/commit/da652cf3cca789d053c06fb069b25e37d95848da#diff-ae6e1415a6620b312668ab17564ad6e0)

looks like it, yes.

ipa-led, 2019-07-02:
Method applied is described here: https://github.com/ros-industrial/universal_robot/pull/360

gavanderhoorn, 2019-07-03:
did you make any progress with this @rtonnaer?

gavanderhoorn, 2019-07-09:
@rtonnaer: I'll unassign this for now. We'll keep the issue around for someone else to work on.

Thanks @ipa-led for some initial investigation.

Briancbn, 2020-07-06:
Hi, I am familiar with blender, I will tackling this in wrid20