ros-industrial/universal_robot #427: Gazebo simulation not working

davizinho5, 2019-05-30:
Hello, 

I'm trying to use the universal packages in Ubuntu 18.04 and ROS-melodic. 

I installed the packages from source and downloaded all the dependencies manually. 

When I launched the Gazebo simulation, first it complained about not having control parameters in the PIDs. I solved it following the last comment in issue #397  , adding some gains. 

I don't see any error in the launch log, only the controller manager and spawner "finishing cleanly" make me make me doubt if this message is correct.  You can see it below:

`[ros_control_controller_manager-9] process has finished cleanly
log file: /home/amper/.ros/log/607db2d0-82e0-11e9-bdb8-f07bcb7ff7ac/ros_control_controller_manager-9*.log
Loaded 'arm_controller'
Loaded 'joint_state_controller'
Started ['arm_controller'] successfully
Started ['joint_state_controller'] successfully
[joint_state_controller_spawner-7] process has finished cleanly
log file: /home/amper/.ros/log/607db2d0-82e0-11e9-bdb8-f07bcb7ff7ac/joint_state_controller_spawner-7*.log
[arm_controller_spawner-8] process has finished cleanly
log file: /home/amper/.ros/log/607db2d0-82e0-11e9-bdb8-f07bcb7ff7ac/arm_controller_spawner-8*.log`

Then, the arm looks weird to me:
[image]

Finally, when I try to move it by commanding a trajectory, it looks like it tries to moves but it cannot:

````
rostopic pub /arm_controller/command trajectory_msgs/JointTrajectory "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
joint_names:
 ['shoulder_pan_joint', 'shoulder_lift_joint', 'elbow_joint', 'wrist_1_joint', 'wrist_2_joint', 'wrist_3_joint']
points:
- positions: [2,0,0,0,0,0]
  velocities: []
  accelerations: []
  effort: []
  time_from_start: {secs: 2, nsecs: 0}" 
````

Anyone knows why this happens? 
Could it be that the gains are not appropiate? How can I know when they are?
If it is because the controllers die, can you imagine why?

Best,

miguelprada, 2019-05-30:
> Could it be that the gains are not appropiate? How can I know when they are?

It could be, yes. What happens if you revert your fix to the the PID error and try to move the robot?

> If it is because the controllers die, can you imagine why?

It's not the controllers that are dying, but the process that takes care of spawning them. That's perfectly normal.

davizinho5, 2019-05-31:
Hi, thanks for the answer.

I reverted it and it works. I'm going to point this out in #397 because the answers are quite confusing.  
My problem was that yesterday I tried a movement that made the arm collision with the floor, and I understood that a similar problem was happening. 

However, following the instructions in here (http://wiki.ros.org/ur_gazebo) for working with a simulated robot in gazebo (with limited joint position) and planning with moveit, I still find problems. 

This first launch works fine:
`roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true`

Then, the second launch opens the rviz + moveit interface but the robot is in a very strange pose. Besides, when I try to move the goal arrows, it has a strange bahaviour, like if all the links where placed in the same point. 
`roslaunch ur5_moveit_config moveit_rviz.launch config:=true`

[image]

Do you know why this might happen?

gavanderhoorn, 2019-05-31:
> Then, the second launch opens the rviz + moveit interface but the robot is in a very strange pose.

See #374, specifically: https://github.com/ros-industrial/universal_robot/issues/374#issuecomment-411025881.

---

off-topic: I really wish this would be fixed upstream, as it's getting annoying having to support users with this every time.

davizinho5, 2019-05-31:
Solved, thanks a lot. 
Sorry for not finding the solution in the issues before asking.

closed 2019-05-31 by davizinho5

gavanderhoorn, 2019-05-31:
My comment was not directed at you. It was more a general comment on the fact that the issue with `urdfdom` and Melodic/Ubuntu Bionic has been around for almost 2 years now and *still* hasn't been resolved.

davizinho5, 2019-05-31:
No worries, I didn't take it personally. 
But I can understand that sometimes the solution should be on the side of the ones that are making the questions.

ar-i-enterprise, 2020-05-05:
how did you fixed ? please i need help with this error..

swagatk, 2020-09-18:
@davizinho5 - HI, can you please let me know how did you figure out the right PID values for the UR5 simulator? How did you fix the PID error? 

> I reverted it and it works. I'm going to point this out in #397 because the answers are quite confusing. 

Can you please elaborate on this - regarding what values did you use. I added the following content to my 'ur_gazebo/controller/arm_controller_ur5.yaml'

`
gazebo_ros_control:
  pid_gains:
    shoulder_pan_joint: {p: 100.0, i: 0.01, d: 10.0}
    shoulder_lift_joint: {p: 100.0, i: 0.01, d: 10.0} 
    elbow_joint: {p: 100.0, i: 0.01, d: 10.0}  
    wrist_1_joint: {p: 100.0, i: 0.01, d: 10.0}
    wrist_2_joint: {p: 100.0, i: 0.01, d: 10.0}
    wrist_3_joint: {p: 100.0, i: 0.01, d: 10.0}
`
It does not show me any error but I am not able to move the arm by publishing at the rostopic, exactly as you mentioned in your post. But, I could not understand what did you fix the problem.

I will be grateful for your help.

davizinho5, 2020-09-19:
Hi, 

I did not use any specific gain. 
This is the message that explains what I really had to do, it had nothing to do with the gains:

https://github.com/ros-industrial/universal_robot/issues/427#issuecomment-497661952

swagatk, 2020-09-19:
So basically, you are using moveit to control the robot arm.

Thanks. This helps. I was spending a lot of time tuning the gains.

Regards,
Swagat